borglab/gtsam #184: Request/Question: boost::shared_ptr -> std C++11 versions?

jlblancoc, 2019-12-07:
## Feature

I wonder whether you might be interested in porting all boost::shared_ptr to standard C++11 std::shared_ptr and std::make_shared. 
Probably the only downside is that it might break some user code, so such a change should bump a (minor?) release version.
I could do a PR to do the port and fix all examples and tests, if you are interested. 

## Motivation

Use standard C++11 features, since C++11 is already a requirement...
Also, use std::make_shared a feature (boost::make_shared) that is right now not used.

dellaert, 2019-12-07:
GTSAM is used by many companies on a variety of platforms, including embedded ones where compiler versions might be lagging. This is why C++17 and std:shared_ptr are not suited for 4.X versions...

dellaert, 2019-12-07:
BTW, I know we already demand c+11, but “breaking some user code” is actually “breaking a *lot* of user code”

jlblancoc, 2019-12-07:
Alright, it makes sense then to remain with status quo :-)

closed 2019-12-07 by jlblancoc