borglab/gtsam #191: gtsam_eigency finds wrong python interpreter

tim-fan, 2019-12-13:
## Description

Trying to build the cython wrapper, with a python3 virtualenv activated, invoking cmake throws the error `ImportError: No module named 'Cython'`

Cython is installed in the virtualenv.

Last few lines of cmake output:
```
-- Found PythonInterp: /export/bulk/local-home/tfanselo/projects/20191213_gtsam_build/venv3/bin/python (found version "3.5.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found version "3.5.2") 
-- Found Cython: /export/bulk/local-home/tfanselo/projects/20191213_gtsam_build/venv3/lib/python3.5/site-packages/Cython (found suitable version "0.29.14", minimum required is "0.25.2") 
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'Cython'
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Cython (missing: CYTHON_PATH CYTHON_EXECUTABLE) (found
  suitable version "0.29.14", minimum required is "0.25.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindCython.cmake:72 (find_package_handle_standard_args)
  cmake/GtsamCythonWrap.cmake:16 (find_package)
  cython/gtsam_eigency/CMakeLists.txt:1 (include)
```

From that, it appears that python is found twice - first the virtualenv python and cython is found, then system python is found, and the cython error is thrown.

Digging into the issue a little, it looks like `./cmake/GtsamCythonWrap.cmake` is included twice, first from `./cython/CMakeLists.txt`, then from `./cython/gtsam_eigency/CMakeLists.txt`. In the second include the system python is found, and hence cython is not found.

If I comment out `include(GtsamCythonWrap)` on first line of `./cython/gtsam_eigency/CMakeLists.txt`, then the issue goes away and I can invoke cmake without any errors.

My cmake understanding is pretty thin, so I'm not sure why this happens or what a suitable fix would be.

## Steps to reproduce

```
cd /tmp/
python3 -m venv venv
source ./venv/bin/activate
git clone https://github.com/borglab/gtsam.git
pip install -r gtsam/cython/requirements.txt
mkdir gtsam/build && cd gtsam/build
cmake -DGTSAM_INSTALL_CYTHON_TOOLBOX=on ..
```

## Expected behavior

No cmake errors

## Environment

```
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
```

## Additional information

varunagrawal, 2020-07-07:
Hi @tim-fan sorry for the delay in responding.

I did an intense rewrite of the CMake for the wrapper recently so I should be able to resolve this. I'll get back to you once I have a solution in place.

varunagrawal, 2020-07-12:
So I am unable to recreate this bug on my system, and CMake shows me the virtualenv versions of Python and Cython.
My guess is this issue got resolved when we cleaned up the Cython wrapper cmake files over the last few months.

Closing this issue. If you still believe this issue exists @tim-fan, please feel free to reopen.

closed 2020-07-12 by varunagrawal