borglab/gtsam #196: ?? () from /usr/lib/x86_64-linux-gnu/libtbb.so.2

narutojxl, 2019-12-19:
## Description
When using  gtsam-4.0.0-alpha2 and ceres-solver-1.14.0,  gtsam::ISAM2::update() can not normal execute. When comment  variable about `ceres::Solver::Summary summary;` the update() function can work well. My Eigen3 is located in  /usr/include/eigen3. What ceres version can work well with 
gtsam-4.0.0-alpha2?

## Steps to reproduce

1.  Build gtsam-4.0.0-alpha2 with default cmake params: `mkdir build && cd build && cmake .. && make && sudo make install `
2. Build ceres-1.14.0  with default cmake params like above.
3. When  define a  `ceres::Solver::Summary summary;` in code like the following, the code is stuck  in executing `isam->update(gtSAMgraph, initialEstimate); `
> 	
    ISAM2* isam;
	ISAM2Params parameters;
	parameters.relinearizeThreshold = 0.001; 
	parameters.evaluateNonlinearError = true;
	parameters.relinearizeSkip = 1;

	isam = new ISAM2(parameters);

	NonlinearFactorGraph gtSAMgraph;
	//do something with gtSAMgraph

	Values initialEstimate;
	//do something with initialEstimate

	ceres::Solver::Summary summary;

	isam->update(gtSAMgraph, initialEstimate); 

4.  The gdb output is following:
>   	
       Thread 1 "mapOptmization" received signal SIGINT, Interrupt.
	0x00000000008baa56 in Eigen::internal::gemm_pack_lhs<double, long, Eigen::internal::const_blas_data_mapper<double, long, 1>, 4, 2, 1, false, false>::operator()(double*, Eigen::internal::const_blas_data_mapper<double, long, 1> const&, long, long, long, long) [clone .constprop.1259] ()
	  Id   Target Id         Frame 
	* 1    Thread 0x7ffff7f80900 (LWP 28939) "mapOptmization" 0x00000000008baa56 in Eigen::internal::gemm_pack_lhs<double, long, Eigen::internal::const_blas_data_mapper<double, long, 1>, 4, 2, 1, false, false>::operator()(double*, Eigen::internal::const_blas_data_mapper<double, long, 1> const&, long, long, long, long) [clone .constprop.1259] ()
	  2    Thread 0x7fffe6b70700 (LWP 28951) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  3    Thread 0x7fffe636f700 (LWP 28952) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  4    Thread 0x7fffe3b6e700 (LWP 28953) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  5    Thread 0x7fffe136d700 (LWP 28954) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  6    Thread 0x7fffdcb6c700 (LWP 28955) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  7    Thread 0x7fffdc36b700 (LWP 28956) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  8    Thread 0x7fffd7b6a700 (LWP 28957) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  9    Thread 0x7fffd5369700 (LWP 28958) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  10   Thread 0x7fffd2b68700 (LWP 28959) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  11   Thread 0x7fffd0367700 (LWP 28960) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  12   Thread 0x7fffcdb66700 (LWP 28961) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  13   Thread 0x7fffcaf53700 (LWP 28963) "mapOptmization" 0x00007ffff1873a13 in epoll_wait () at ../sysdeps/unix/syscall-template.S:84
	  14   Thread 0x7fffca752700 (LWP 28964) "mapOptmization" 0x00007ffff186774d in poll () at ../sysdeps/unix/syscall-template.S:84
	  15   Thread 0x7fffc9f51700 (LWP 28965) "mapOptmization" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
	  16   Thread 0x7fffc9750700 (LWP 28977) "mapOptmization" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
	  17   Thread 0x7fffc8f4f700 (LWP 29018) "mapOptmization" 0x00007ffff3f48c1d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
	[Switching to thread 1 (Thread 0x7ffff7f80900 (LWP 28939))]
	#0  0x00000000008baa56 in Eigen::internal::gemm_pack_lhs<double, long, Eigen::internal::const_blas_data_mapper<double, long, 1>, 4, 2, 1, false, false>::operator()(double*, Eigen::internal::const_blas_data_mapper<double, long, 1> const&, long, long, long, long) [clone .constprop.1259] ()
	#0  0x00000000008baa56 in Eigen::internal::gemm_pack_lhs<double, long, Eigen::internal::const_blas_data_mapper<double, long, 1>, 4, 2, 1, false, false>::operator()(double*, Eigen::internal::const_blas_data_mapper<double, long, 1> const&, long, long, long, long) [clone .constprop.1259] ()
	#1  0x00000000008c084b in Eigen::internal::general_matrix_matrix_product<long, double, 1, false, double, 0, false, 0>::run(long, long, long, double const*, long, double const*, long, double*, long, double, Eigen::internal::level3_blocking<double, double>&, Eigen::internal::GemmParallelInfo<long>*) ()
	#2  0x00007ffff38a442b in gtsam::JacobianFactor::updateHessian(gtsam::FastVector<unsigned long> const&, gtsam::SymmetricBlockMatrix*) const () from /usr/local/lib/libgtsam.so.4
	#3  0x00007ffff388c9f7 in gtsam::HessianFactor::HessianFactor(gtsam::GaussianFactorGraph const&, boost::optional<gtsam::Scatter const&>) () from /usr/local/lib/libgtsam.so.4
	#4  0x00007ffff38952c5 in boost::detail::sp_if_not_array<gtsam::HessianFactor>::type boost::make_shared<gtsam::HessianFactor, gtsam::GaussianFactorGraph const&, gtsam::Scatter&>(gtsam::GaussianFactorGraph const&, gtsam::Scatter&) () from /usr/local/lib/libgtsam.so.4
	#5  0x00007ffff388caff in gtsam::EliminateCholesky(gtsam::GaussianFactorGraph const&, gtsam::Ordering const&) () from /usr/local/lib/libgtsam.so.4
	#6  0x00007ffff388ccab in gtsam::EliminatePreferCholesky(gtsam::GaussianFactorGraph const&, gtsam::Ordering const&) () from /usr/local/lib/libgtsam.so.4
	#7  0x00007ffff3855153 in boost::detail::function::function_invoker2<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (*)(gtsam::GaussianFactorGraph const&, gtsam::Ordering const&), std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> >, gtsam::GaussianFactorGraph const&, gtsam::Ordering const&>::invoke(boost::detail::function::function_buffer&, gtsam::GaussianFactorGraph const&, gtsam::Ordering const&) () from /usr/local/lib/libgtsam.so.4
	#8  0x00007ffff3919f4e in gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >::EliminationPostOrderVisitor::operator()(boost::shared_ptr<gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster> const&, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >&) () from /usr/local/lib/libgtsam.so.4
	#9  0x00007ffff391ac45 in gtsam::treeTraversal::internal::PreOrderTask<gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> > (boost::shared_ptr<gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster> const&, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >&), gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >::EliminationPostOrderVisitor>::execute() () from /usr/local/lib/libgtsam.so.4
	#10 0x00007ffff20f4fdd in ?? () from /usr/lib/x86_64-linux-gnu/libtbb.so.2
	#11 0x00007ffff3919c8c in gtsam::treeTraversal::internal::RootTask<gtsam::FastVector<boost::shared_ptr<gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster> >, gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> >, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::GaussianFactorGraph> > (boost::shared_ptr<gtsam::ClusterTree<gtsam::GaussianFactorGraph>::Cluster> const&, gtsam::EliminationData<gtsam::EliminatableClusterTree<gtsam::ISAM2BayesTree, gtsam::Ga [... cut ...]
	#17 0x0000000000713c4a in mapOptimization::saveKeyFramesAndFactor (this=0x7fffffff0580) at /home/jxl/jxl_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:2166
	#18 0x000000000071557d in mapOptimization::run (this=0x7fffffff0580) at /home/jxl/jxl_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:2309
	#19 0x00000000006f68ed in main (argc=1, argv=0x7fffffff3c38) at /home/jxl/jxl_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:2337
	Quit

   

## Environment

Linux dell 4.15.0-54-generic  #58~16.04.1-Ubuntu SMP Mon Jun 24 13:21:41 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
gcc-5.4.0, g++-5.4.0, cmake-3.16.0,  ubuntu16.04

closed 2019-12-21 by narutojxl

zeng-xiangwei, 2020-06-16:
@narutojxl Hello, I also meet this issue. Have you solved the problem? Thanks.

narutojxl, 2020-06-16:
Hi @zeng-xiangwei, no. :(  I give up using ceres with gtsam eventually, through many attempts failed. I think maybe it is a gtsam-4.0.0-alpha2 bug or something else. Maybe you can test with the newer gtsam version, i didn't do that test at that time.

zeng-xiangwei, 2020-06-16:
@narutojxl Thank you very much.

zeng-xiangwei, 2020-06-16:
@narutojxl  I test with the newer gtsam version(release: tag 4.0.2), the update() function can work well. Thanks again!!!

narutojxl, 2020-06-16:
Thanks a lot ! @zeng-xiangwei

yxllnSLAM, 2022-02-22:
@zeng-xiangwei @narutojxl 两位好，我在4.0.2的分支上依然会出现问题，后来你们有尝试吗

zeng-xiangwei, 2022-02-22:
> @zeng-xiangwei @narutojxl 两位好，我在4.0.2的分支上依然会出现问题，后来你们有尝试吗

@yxllnSLAM 你好，我在ubuntu20.04上用的ceres1.14.0+gtsam4.0.3，把ceres库重新安装为共享库（.so）就没有这个问题了（安装ceres的时候加一句`cmake -DBUILD_SHARED_LIBS=ON ..`）。
我目前还不清楚为什么换成 .so 就好用了。

yxllnSLAM, 2022-02-22:
> > @zeng-xiangwei @narutojxl 两位好，我在4.0.2的分支上依然会出现问题，后来你们有尝试吗
> 
> @yxllnSLAM 你好，我在ubuntu20.04上用的ceres1.14.0+gtsam4.0.3，把ceres库重新安装为共享库（.so）就没有这个问题了（安装ceres的时候加一句`cmake -DBUILD_SHARED_LIBS=ON ..`）。 我目前还不清楚为什么换成 .so 就好用了。

@zeng-xiangwei 好的好的，多谢分享