borglab/gtsam #203: errors about “relocation R_X86_64_32S against `_ZTVN5boost7archive6detail14basic_iarchiveE' ” occur when make！

zengletian1491, 2019-12-30:
The following operations are performed on Ubuntu 16.04 LTS. The commands for the installation of GTSAM are listed in the following:
**$ git clone --branch 4.0.0-alpha2 https://github.com/borglab/gtsam.git gtsam-4.0.0-alpha2
$ cd gtsam-4.0.0-alpha2
$ mkdir build
$ cd build
$ cmake -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_UNSTABLE=OFF ..
$ make**

However, errors occur as follows:

**[100%] Linking CXX shared library libgtsam.so
/usr/bin/ld: /usr/local/lib/libboost_serialization.a(basic_iarchive.o): relocation R_X86_64_32S against `_ZTVN5boost7archive6detail14basic_iarchiveE' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libboost_serialization.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
gtsam/CMakeFiles/gtsam.dir/build.make:1868: recipe for target 'gtsam/libgtsam.so.4.0.0' failed
make[2]: *** [gtsam/libgtsam.so.4.0.0] Error 1
CMakeFiles/Makefile2:410: recipe for target 'gtsam/CMakeFiles/gtsam.dir/all' failed
make[1]: *** [gtsam/CMakeFiles/gtsam.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2**

Could you please give me some suggestions?Thanks!

ProfFan, 2019-12-30:
Have you tried latest `develop` branch?

zengletian1491, 2019-12-31:
> Have you tried latest `develop` branch?

Thanks！The develop branch works well！

ProfFan, 2019-12-31:
Feel free to open another issue if you have problem with develop.

closed 2019-12-31 by ProfFan

zengletian1491, 2019-12-31:
> Feel free to open another issue if you have problem with develop.

OK！Thanks very much！