borglab/gtsam #204: Modern wrapper plan

dellaert, 2020-01-02:
Desired end-state: python scripts generate both python and MATLAB wrappers. Because there is an API change in python, this will bump our version to 4.1.

Plan, in @ProfFan  fork:

1. Add travis CI for cython wrapper
2. Add pybind wrapper version, also with CI support (purely additive, duplicates all python scripts/tests into gtsam_py, includes GTSAM changes)
3. Create PR 1 with both these changes to gtsam
4. Create packaging so we can pip install, PR 2 to gtsam
5. Do PR to master with 4.0.4 -> needed for CS 3630
6. Switch MATLAB wrapper to be created by python, includes Matthew unit tests, PR 3
7. Remove cython wrapper and old c++ wrap, PR 4 to gtsam (only PR with file deletions)
8. Do PR to master with 4.1

ProfFan, 2020-01-06:
@dellaert Now gtsam works in Colab with some patches :)

Demo: https://colab.research.google.com/drive/1j2HgxVUDUfQEseLmke4Tk_LKiyJ87QjQ

One blocker: pypi got a file size limit of 60MB, but GTSAM packaged is 272M... I filed a request for increase the quota, in the meantime can load from Google Drive or an URL instead.

ProfFan, 2020-01-06:
Code to build the wheels:

https://github.com/ProfFan/gtsam-manylinux-build

ProfFan, 2020-01-07:
@dellaert Now GTSAM lives on PyPI with a size of 7.3MB :)

I changed the Colab notebook to just use `pip install gtsam` :smile:

dellaert, 2020-01-07:
This is *AMAZING*

dellaert, 2020-01-07:
How did you get it down to 7.3MB??

ProfFan, 2020-01-07:
I stripped out the debug symbols and used a custom boost build instead of the system provided package...

ProfFan, 2020-01-19:
@dellaert Note that macOS build is complete and tested (Python 3.7).

dellaert, 2020-01-19:
> @dellaert Note that macOS build is complete and tested (Python 3.7).

Does that mean we can now move on to "Add pybind wrapper version, also with CI support (purely additive, duplicates all python scripts/tests into gtsam_py, includes GTSAM changes)" ?

ProfFan, 2020-01-19:
Yes.

dellaert, 2020-02-09:
@ProfFan Did you add Travis CI for (both) wrappers in your branch yet? I think that's crucial before we can merge into develop.

dellaert, 2020-02-09:
@ProfFan I checked out the branch but confused about flags. There is `GTSAM_INSTALL_CYTHON_TOOLBOX` and `GTSAM_BUILD_PYTHON`. Should the latter be `GTSAM_INSTALL_PYTHON_TOOLBOX` ?

ProfFan, 2020-02-09:
I kept them this way because that is what duy used. After all, I think the first `INSTALL` should be renamed to `GTSAM_BUILD_CYTHON_WRAPPER` as this flags actually does not control the installation of the wrapper. It only builds the wrapper.

ProfFan, 2020-02-09:
@dellaert Yes, but I haven't added the actual tests, just pure compile (and package to wheel). I will add it really quick.

dellaert, 2020-02-09:
I agree, but I propose we are consistent with the (slightly wrong) past, and use `GTSAM_INSTALL_PYTHON_TOOLBOX`

ProfFan, 2020-02-10:
@dellaert Got it. I'll go that way  :)

chrisbeall, 2020-03-25:
What is the current state of the pybind wrapper? Is it ready for testing in `feature/new_wrapper`, or are there still significant missing pieces? We are very interested in trying this at Nuro :-)

dellaert, 2020-03-25:
@ProfFan good idea: please give status update and update TODO list above.

ProfFan, 2020-03-25:
@chrisbeall @dellaert It is in a *kind-of* final state-except it is blocked by the Travis timeouts because the generated `gtsam.cpp` is too large. It builds (and tests) nicely on GitHub Actions, and that is the reason behind #259 :)

dellaert, 2020-03-25:
Can you update checkmarks above? What's missing ?

dellaert, 2020-03-25:
@chrisbeall consider commenting/approving #259 :-)

dellaert, 2020-08-19:
@ProfFan two remaining items after mega-merge:

- [ ] Can we now make typedef default and simply remove conditional code?
- [ ]  After that's done, please tag a release 4.1rc, and create a corresponding python wheel?

BioGeek, 2021-03-19:
Small note, in the [Colab notebook](https://colab.research.google.com/drive/1j2HgxVUDUfQEseLmke4Tk_LKiyJ87QjQ) by @ProfFan  [mentioned earlier](https://github.com/borglab/gtsam/issues/204#issuecomment-571344287), you have to change `gtsam.noiseModel_Diagonal.Sigmas`  to `gtsam.noiseModel.Diagonal.Sigmas` to get it to run correctly.

closed 2021-07-14 by ProfFan