borglab/gtsam #205: Inconsistent in-code comments of variable order in Pose3 noiseModel.

ptrmu, 2020-01-04:
imuFactorExample2.cpp contains the following lines:

```
  // Add a prior on pose x0. This indirectly specifies where the origin is.
  // 0.1 rad std on roll, pitch, yaw, 30cm std on x,y,z.
  auto noise = noiseModel::Diagonal::Sigmas(
      (Vector(6) << Vector3::Constant(0.1), Vector3::Constant(0.3)).finished());
  newgraph.push_back(PriorFactor<Pose3>(X(0), pose_0, noise));
```

VisualISAM2Example.cpp contains the following lines:

```
      // Add a prior on pose x0, 30cm std on x,y,z and 0.1 rad on roll,pitch,yaw
      static auto kPosePrior = noiseModel::Diagonal::Sigmas(
          (Vector(6) << Vector3::Constant(0.3), Vector3::Constant(0.1))
              .finished());
      graph.emplace_shared<PriorFactor<Pose3> >(Symbol('x', 0), poses[0],
                                                kPosePrior);
```

I believe the ordering in the first example is correct but the second is more common in the code base (.cpp, .py, .m). This inconsistency adds to the confusion caused by the fact that Pose2 and Pose3 have different spatial and rotation ordering.

dellaert, 2020-01-04:
Yeah, the first one is right, second one is wrong :-( I think someone fixed the first one but forgot to fix the copy/pasted instances. Wanna try a PR?

ptrmu, 2020-01-05:
Sure. Will do a PR.

ProfFan, 2020-01-05:
Thank you very much!

closed 2020-01-05 by ProfFan