borglab/gtsam #208: TBB Deprecation Warnings

acxz, 2020-01-11:
## Description

TBB has been updated and now when building gtsam the build output is cluttered with deprecation warnings.
```
[ 31%] Building CXX object gtsam/CMakeFiles/gtsam.dir/base/debug.cpp.o
In file included from /home/acxz/vcs/git/github/borglab/gtsam/gtsam/base/types.h:31,
                 from /home/acxz/vcs/git/github/borglab/gtsam/gtsam/global_includes.h:22,
                 from /home/acxz/vcs/git/github/borglab/gtsam/gtsam/base/debug.h:20,
                 from /home/acxz/vcs/git/github/borglab/gtsam/gtsam/base/debug.cpp:19:
/usr/include/tbb/task_scheduler_init.h:21:154: note: #pragma message: TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                  ^

In file included from /home/acxz/vcs/git/github/borglab/gtsam/gtsam/base/debug.cpp:23:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
```

## Steps to reproduce

1. git clone https://github.com/borglab/gtsam
2. mkdir build && cd build && cmake .. && make

## Expected behavior

No TBB deprecation warnings.

## Environment

OS: ArchLinux
GTSAM version: latest `develop` branch (https://github.com/borglab/gtsam/commit/7ebfdadb2840949baac5d3b096459c5168d73d00) at time of posting
TBB version: 2020.0

## Additional information

acxz, 2020-04-05:
Focal Fossa (the latest LTS version of Ubuntu) will be released on Apr 23rd [see](https://wiki.ubuntu.com/FocalFossa/ReleaseSchedule)

The TBB version it will ship with is 2020.1 [see](https://packages.ubuntu.com/focal/libtbb2)
This will cause warnings that showed up in other distros like Arch to show up in Ubuntu.

If #209 still requires a longer time to review, can we bandaid the issue by suppressing the TBB warnings in the GTSAM code.

ruffsl, 2020-05-05:
These deprecation warnings are getting spamy in my Ubuntu 20.04 build logs. It'd be nice to see #209 merged so we don't have to monkey patch other cmake projects that have drowned out warnings.

dellaert, 2020-05-06:
> These deprecation warnings are getting spamy in my Ubuntu 20.04 build logs. It'd be nice to see #209 merged so we don't have to monkey patch other cmake projects that have drowned out warnings.

@ruffsl can you confirm #209 works for you as expected, and both TBB and non-TBB compile and run as expected?

acxz, 2020-05-07:
@ruffsl the warnings up to TBB 2020.1 should be gone, there are still deprecation warnings with TBB 2020.2 that need to be addressed. Particularly the deprecation of `task.h` which affects quite a few classes and files in GTSAM.

ruffsl, 2020-05-07:
Thanks @acxz , the downstream build logs are a quieter:

```
--- stderr: gtsam
/opt/ros/underlay_ws/src/borglab/gtsam/gtsam/3rdparty/CCOLAMD/Source/ccolamd.c: In function ‘ccolamd2’:
/opt/ros/underlay_ws/src/borglab/gtsam/gtsam/3rdparty/CCOLAMD/Source/ccolamd.c:1563:9: warning: variable ‘cs’ set but not used [-Wunused-but-set-variable]
 1563 |     Int cs ;
      |         ^~
---
```

Although, I only tested it with TBB enabled as thats what my project also needs to run.

acxz, 2020-05-07:
Glad to hear that, that specific warning is due to CCOLAMD which is a 3rd party codebase.

varunagrawal, 2020-07-07:
@acxz is this resolved?

acxz, 2020-07-07:
Nope not yet. see #336 for a wip solution

acxz, 2021-06-17:
Update:

Looks like intel released a better migration guide: https://docs.oneapi.com/versions/latest/onetbb/tbb_userguide/Migration_Guide/Task_API.html

Based on that I updated my attempt: https://github.com/acxz/gtsam/commit/825fe742528afce354778f9466a6bfbd8e65971f
Tests success has risen from 75% to 85% from my previous attempt. (Pardon the print statements)

PR #800 (2021-06-23): replace deprecated tbb functionality
resolves #208 continuation of #336 

I was able to replace the creation of the old api tasks with functors being passed into task_groups.

Since this is a change to core functionality, I urge people to review and check unit tests thoroughly.
closed 2021-06-23 by varunagrawal