borglab/gtsam #213: IMUFactor example noise/bias units

Jason420Git, 2020-01-17:
Hi, I am trying to use GTSAM to create a system using IMU and GPS measurements together in a factor graph.

I was following the example IMUFactor. It has the following values which were being computed from the sensor specifications
```

// We use the sensor specs to build the noise model for the IMU factor.
  double accel_noise_sigma = 0.0003924;
  double gyro_noise_sigma = 0.000205689024915;
  double accel_bias_rw_sigma = 0.004905;
  double gyro_bias_rw_sigma = 0.000001454441043;
```

However there are no units attached to these. I am trying to better model the IMU I am using which is Bosch BMI160. I cannot find any values which resemble these. 

https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000.pdf

Any help in adapting the sample to use this IMU would be of great help!

jingyibo123, 2020-04-09:
Did you solve the myth @Jason420Git

mikesheffler, 2020-04-10:
For what it's worth, [here's a relevant disucssion](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/gtsam-users/o9brW87ZjZ0/dWOigRcjBAAJl) from the mailing list last fall. A quick summary could be, "Read the IMU noise model wiki page over at the ETHZ Kalibr project", and, "converting between different units is hard."

If you take a look at that datasheet from Bosch, two numbers that jump out at me that will be relevant for you are, 'Output noise' (top of page 9) `ug/sqrt(Hz)` and 'Output noise' (near top of page 10) `deg/s/sqrt(Hz)` those are noises. To figure out how the noise _rate_ changes, you pretty much need an Allan variance plot, so you

a) get that from the manufacturer
b) do the tests yourself
c) guess

with low-quality IMUs, guessing is not insane. The manufacturer will usually include a 'bias stability' or 'bias instability' number, and once you understand how Allan variance plots work, you can use that bias instability as an upper bound for the gyro random walk and accel random walk values.

Good luck!

varunagrawal, 2020-07-09:
So I have made some progress on this, but not enough to have a satisfactory answer for everyone. Unfortunately, I have a deadline coming up next week so I am going to have to push this back until then.

I am assigning this to myself though and you can rest assured I will figure this out. :)

mikesheffler, 2020-07-11:
So, I thought that the submitter was trying to find appropriate numerical values; if they just want the units, I would be shocked if they do not correspond to the [IMU  noise model at the Kalibr wiki](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model). that is:
  * `accel_noise_sigma` &#8594; `accelerometer_noise_density`
  * `gyro_noise_sigma` &#8594; `gyroscope_noise_density`
  * `accel_bias_rw_sigma` &#8594; `accelerometer_random_walk`
  * `gyro_bias_rw_sigma` &#8594; `gyroscope_random_walk`

This is strongly implied by [Forster et al., TRO '16](http://rpg.ifi.uzh.ch/docs/TRO16_forster.pdf) in footnote 2 on page 10, where the IMU parameters for the simulations are listed and match. Also, Kalibr is mentioned in the paper, although not explicitly in the context of calibrating IMU measurements.

As @dellaert said in the e-mail chain I linked above, Forster and Luca Carlone (cc @lucacarlone ) are probably the only ones that know for sure.

lucacarlone, 2020-07-16:
@mikesheffler : correct! the variables accel_noise_sigma, gyro_noise_sigma, accel_bias_rw_sigma,  gyro_bias_rw_sigma are the continuous-time noise densities of the imu noise and the random walks (the discretization happens inside the factors). Therefore the units are the same as the ones mentioned in the Kalibr wiki, except a couple of (minor) typos in the Kalibr github: https://github.com/ethz-asl/kalibr/issues/354   
Let me know if you have further questions!

varunagrawal, 2020-08-11:
So after spending a lot of time deep in the trenches with the code, and even understanding noise derivation via Allan Deviation plots, I can confirm what @lucacarlone already mentioned.

The units are:

- `gyro_noise_sigma`: `rad/s/√Hz` (Gyro white noise)
- `accel_noise_sigma`: `m/s²/√Hz` (Accelerometer white noise)
- `gyro_bias_rw_sigma`: `rad/s` or `rad √Hz/s` (Gyro bias stability or random walk)
- `accel_bias_rw_sigma`: `m/s²` or `m √Hz/s²` (Accelerometer bias stability or random walk)

@dellaert please recommend where should I put this info?

closed 2020-12-02 by varunagrawal

zzodo, 2021-01-07:
Love you guys! This helped me a lot :+1:

SubMishMar, 2021-03-07:
Can bias stability and random walk used interchangeably? For example if my IMU dataset says that the bias stability is 0.0004 m/s^{2} then can I directly plug in the value where the expected unit is  m √Hz/s^{2} ? 

I am a little confused about this.

mikesheffler, 2021-03-08:
> Or do I have to multiply the √ of my imu frequency to it.

Yes, there is a conversion necessary. I don't believe the conversion involves the IMU frequency.

Take a look at [this post on the Analog Devices forum](https://ez.analog.com/mems/f/q-a/88768/adxrs453-angular-random-walk-and-bias-stability) . That person is talking about a gyroscope instead of an accelerometer, but their question is fundamentally similar to yours. Look at  NevadaMark's answer. There, he's talking about values related to the (root) Allan variance curve, which (the units from the curve) are specified in deg / s. He's then converting those numbers to units to talk about values in deg / sqrt(hr) and deg / hr. Your problem doesn't involve units of hours, but note how he uses the sqrt in his conversion because of the units of the angle random walk vs the units of the bias stability.

mintar, 2021-11-26:
> Can bias stability and random walk used interchangeably?

No.

### IEEE Standard 952-2020

Source: [IEEE Standard 952-2020](https://doi.org/10.1109/IEEESTD.2021.9353434) for Specifying and Testing Single-Axis Interferometric Fiber Optic Gyros (page 18f)

Allan variance component                                  | unit
----------------------------------------------------------|---------------------
gyro angle quantization noise coefficient, Q              | ", μrad
gyro angle random walk (rate white noise) coefficient, N  | °/√h, rad/√s
gyro bias instability coefficient, B                      | °/h, rad/s
gyro rate random walk coefficient, K                      | (°/h)/√h, (rad/s)/√h
gyro ramp coefficient, R                                  | (°/h)/h, (rad/s)/s

Unfortunately the IEEE standard is only concerned with gyroscopes, not accelerometers. However, if we substitute `rad/s` with `m/s²`, we get the following units for accelerometers:

Allan variance component                                                      | unit
------------------------------------------------------------------------------|-----------
accelerometer velocity quantization noise coefficient, Q                      | m/s
accelerometer velocity random walk (acceleration white noise) coefficient, N  | (m/s)/√s, (m/s)/√h
accelerometer bias instability coefficient, B                                 | m/s², m/h², mg (= [milli-g-force](https://en.wikipedia.org/wiki/G-force), not milligram)
accelerometer acceleration random walk coefficient, K                         | (m/s²)/√s, (m/s²)/√h
accelerometer ramp coefficient, R                                             | m/s³

These units are consistent with the following table from the [Kalibr wiki](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#the-noise-model-parameters-in-kalibr) (modified by me):

Kalibr Parameter            | Kalibr Wiki Symbol                                              | Allan variance symbol                                              |  Units
--------------------------- | --------------------------------------------------------------- | ------------------------------------------------------------------ | ----------------------------------------------------------------------------------------------
Gyroscope "white noise"     | <img src="https://latex.codecogs.com/svg.latex?{%5Csigma_g}">   | <img src="https://latex.codecogs.com/svg.image?N_g" title="N_g" /> | rad/s * (1/√Hz) = rad/(√s * √s) * √s = rad/√s
Accelerometer "white noise" | <img src="https://latex.codecogs.com/svg.latex?{%5Csigma_a}">   | <img src="https://latex.codecogs.com/svg.image?N_a" title="N_a" /> | m/s² * (1/√Hz) = m/(s * √s * √s) * √s = m/(s * √s) = m/(s * √s) * (√s/√s) = (m/s)/√s
Gyroscope "random walk"     | <img src="https://latex.codecogs.com/svg.latex?{%5Csigma_b_g}"> | <img src="https://latex.codecogs.com/svg.image?K_g" title="K_g" /> | rad/s² * (1/√Hz) = rad/(s * √s * √s) * √s = rad/(s * √s) = rad/(s * √s) * (√s/√s) = (rad/s)/√s
Accelerometer "random walk" | <img src="https://latex.codecogs.com/svg.latex?{%5Csigma_b_a}"> | <img src="https://latex.codecogs.com/svg.image?K_a" title="K_a" /> | m/s³ * (1/√Hz) = m/(s² * √s * √s) * √s = m/(s² * √s) = m/(s² * √s) * (√s/√s) = (m/s²)/√s

*Q* and *R* are irrelevant for most practical applications, so mostly an IMU is only modeled using *N*, *B* and *K* or just *N* and *K* (but never *N* and *B*). I am pretty sure that gtsam only uses *N* and *K*.

Unfortunately, there's a lot of different terms for the same thing:

* *N*: angle random walk, rate white noise, white noise, rate noise density, noise density
* *B*: bias instability, in-run bias stability, flicker noise
* *K*: rate random walk, random walk, bias

For Accelerometers, you have to substitute "angle" → "velocity" and "rate" → "acceleration"

Most IMU datasheets only list *N* and *B*. Also, sometimes they list rms (root mean squared error), which is not what we want; we want the standard deviation.

@mikesheffler wrote:
> So, I thought that the submitter was trying to find appropriate numerical values; if they just want the units, I would be shocked if they do not correspond to the [IMU noise model at the Kalibr [...]

Yes, this is correct.

@varunagrawal wrote:
> The units are:
>
>     * `gyro_noise_sigma`: `rad/s/√Hz` (Gyro white noise)
>     * `accel_noise_sigma`: `m/s²/√Hz` (Accelerometer white noise)
>     * `gyro_bias_rw_sigma`: `rad/s` or `rad √Hz/s` (Gyro [...]

Not quite, at least from what I understand. I think this would be correct:

* `gyro_noise_sigma`: `rad/s/√Hz` (Gyro white noise)
* `accel_noise_sigma`: `m/s²/√Hz` (Accelerometer white noise)
* `gyro_bias_rw_sigma`: ~~`rad/s` or~~ `rad √Hz/s` (Gyro bias ~~stability~~ or random walk)
* `accel_bias_rw_sigma`: ~~`m/s²` or~~ `m √Hz/s²` (Accelerometer bias ~~stability~~ or random walk)

PR #624 (2020-12-02): IMU Parameters Documentation
Fixes #213