borglab/gtsam #222: receive signal SIGSEGV. Segment fault when calling isam.update(graph, initialEstimate);

haotiangu1993, 2020-02-07:
Hi，  I tried to use isam to solve a non-linear optimization problem, but failed and got a segment fault. I debugged the code and found the error was thrown out once I called  isam.update(graph, initialEstimate); and every time when I ran c++ program, it always stop by isam.update(graph, initialEstimate). here is the main part of my program.
```

   
        /**
         * update grsam graph
         */
        if (cloudKeyPoses3D->points.empty()){
            gtSAMgraph.add(PriorFactor<Pose3>(0, Pose3(Rot3::RzRyRx(transformTobeMapped[2], transformTobeMapped[0], transformTobeMapped[1]),
                                                       		 Point3(transformTobeMapped[5], transformTobeMapped[3], transformTobeMapped[4])), priorNoise));
            initialEstimate.insert(0, Pose3(Rot3::RzRyRx(transformTobeMapped[2], transformTobeMapped[0], transformTobeMapped[1]),
                                                  Point3(transformTobeMapped[5], transformTobeMapped[3], transformTobeMapped[4])));
            for (int i = 0; i < 6; ++i)
            	transformLast[i] = transformTobeMapped[i];
        }
        else{
            gtsam::Pose3 poseFrom = Pose3(Rot3::RzRyRx(transformLast[2], transformLast[0], transformLast[1]),
                                                Point3(transformLast[5], transformLast[3], transformLast[4]));
            gtsam::Pose3 poseTo   = Pose3(Rot3::RzRyRx(transformAftMapped[2], transformAftMapped[0], transformAftMapped[1]),
                                                Point3(transformAftMapped[5], transformAftMapped[3], transformAftMapped[4]));
            gtSAMgraph.add(BetweenFactor<Pose3>(cloudKeyPoses3D->points.size()-1, cloudKeyPoses3D->points.size(), poseFrom.between(poseTo), odometryNoise));
            initialEstimate.insert(cloudKeyPoses3D->points.size(), Pose3(Rot3::RzRyRx(transformAftMapped[2], transformAftMapped[0], transformAftMapped[1]),
                                                                     		   Point3(transformAftMapped[5], transformAftMapped[3], transformAftMapped[4])));
        }
        /**
         * update iSAM
         */
        isam.update(gtSAMgraph, initialEstimate);
        isam.update();
        
        gtSAMgraph.resize(0);
        initialEstimate.clear();
```

I used gdb to find the function call frame stack and it showed:
#  in gtsam::NoiseModelFactor::linearize(gtsam::values const & ) const ()from /usr/local/lib/libgtsam.so.4
# in gtsam::(anaonymous namespace)::LinearizeOneFactor::operator()(tbb::blocaked_range<unsigned long> const &)
# in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, gtsam::(anonymous namespace)::LinearizeOneFactor, tbb::auto_partitioner const>::execcute() form
# in ?? () from /usr/lib/x86_64-linux-gun/libtbb.so.2
# in ?? () from /usr/lib/x86_64-linux-gun/libtbb.so.2
# in gtsam::NonlinearFactorGraph::linearize()
# in gtsam::ISAM2::update(gtsam_NonlinwarFactorGraph const&, gtsam::Values const&)

My environment is ubuntu 16.04 with gtsam 4.0.0.  Could anyone please give me some advices? thanks!!

haotiangu1993, 2020-02-07:
```
    NonlinearFactorGraph gtSAMgraph;
    Values initialEstimate;
    Values optimizedEstimate;
    //ISAM2 *isam;
    ISAM2 isam;
    Values isamCurrentEstimate;
```

ProfFan, 2020-02-18:
Please first turn off TBB and build with Debug, thanks.

varunagrawal, 2020-07-07:
@haotiangu1993 can you please share a fully reproducible example? We would like to help you out if we can identify the underlying issue.

varunagrawal, 2021-01-02:
@haotiangu please reopen if this is still an issue for you.

closed 2021-01-02 by varunagrawal