borglab/gtsam #229: [macOS] Travis is actually building with Clang twice

ProfFan, 2020-02-23:
So,

In travis we see macOS builds configured to build with GCC. This is just an illusion. `gcc` on macOS is just a symlink to `clang`, so we are actually building against Clang twice.

Quote from Travis mac GCC build:
```
===============================================================
-- ================  Configuration Options  ======================
--   CMAKE_CXX_COMPILER_ID type     : Clang
--   CMAKE_CXX_COMPILER_VERSION     : 9.1.0.9020039
--   CMake version                  : 3.15.5
--   CMake generator                : Unix Makefiles
--   CMake build tool               : /usr/bin/make
-- Build flags                                               
```

I tested today on GitHub Actions with actual GCC 9 on macOS. It does not work :(

dellaert, 2020-02-23:
1) I guess we should remove from travis then
2) what breaks?

ProfFan, 2020-02-24:
@dellaert 
https://github.com/ProfFan/gtsam/runs/463411363
Quote:
```
Undefined symbols for architecture x86_64:
  "boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::text_iarchive_impl(std::basic_istream<char, std::char_traits<char> >&, unsigned int)", referenced from:
      boost::archive::text_iarchive::text_iarchive(std::basic_istream<char, std::char_traits<char> >&, unsigned int) in SubgraphBuilder.cpp.o
  "boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      void boost::archive::save_access::save_primitive<boost::archive::text_oarchive, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(boost::archive::text_oarchive&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) in SubgraphBuilder.cpp.o
  "boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::text_oarchive_impl(std::basic_ostream<char, std::char_traits<char> >&, unsigned int)", referenced from:
      boost::archive::text_oarchive::text_oarchive(std::basic_ostream<char, std::char_traits<char> >&, unsigned int) in SubgraphBuilder.cpp.o
  "boost::archive::basic_text_iprimitive<std::basic_istream<char, std::char_traits<char> > >::~basic_text_iprimitive()", referenced from:
      boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::~text_iarchive_impl() in SubgraphBuilder.cpp.o
      boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::~text_iarchive_impl() in SubgraphBuilder.cpp.o
  "boost::archive::basic_text_oprimitive<std::basic_ostream<char, std::char_traits<char> > >::~basic_text_oprimitive()", referenced from:
      boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::~text_oarchive_impl() in SubgraphBuilder.cpp.o
      boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::~text_oarchive_impl() in SubgraphBuilder.cpp.o
ld: symbol(s) not found for architecture x86_64
```

ProfFan, 2020-02-24:
Also, when I added more tests to run in the Windows build, the Windows build also failed. The current AppVeyor CI is only running 10% of the test suite.

varunagrawal, 2020-09-12:
Irrelevant since we moved to Actions.

closed 2020-09-12 by varunagrawal