borglab/gtsam #236: [RFC] Switch to GitHub Actions for CI

ProfFan, 2020-03-01:
Recently we have seen a number of CI failures in Travis CI as Travis is not allocating enough time or [not having enough memory](https://github.com/borglab/gtsam/runs/478262154) for GTSAM to build. This blocked some of our PRs.

So, I did a bit of experimenting on GitHub Actions, here are the results:

| Item  |  Travis  | GitHub Actions |
|------|--------|----------|
| Build time | Around 1h | Around 1h |
| (vs.) Test Matrix Size | 18 | 40 |
| Windows |  ❎ | ✅ |
| Linux | ✅ | ✅ |
| macOS | ✅ | ✅ |

Notably, no caching (`ccache`) is used currently with the GitHub Actions CI, so I expect build time to be further reduced if we implement caching. Also, we can reduce the size of the test matrix for non-PR builds, in my experiments if size=20 it can finish in 25 minutes.

My proposal is that we can implement both CI systems at the same time, and allow merging when any one CI passes so we do not get blocked by Travis timeouts.

An example of the current experiment results and setup can be seen at https://github.com/ProfFan/gtsam/actions

Pinging @dellaert @varunagrawal @jlblancoc for comments.

varunagrawal, 2020-03-01:
I've seen a lot of positive reactions on Twitter for GitHub Actions so I fully support this proposal. Moreover, I would prefer to have a docker container prebuilt to be used for CI to save time over there (a lot of Travis errors are due to apt-get timeouts). The container template can also serve as a pseudo dependency guide for anyone wanting to install GTSAM from source or even play around with it.

varunagrawal, 2020-03-21:
I've been playing around with workflows using Github Actions on a fork of GTSAM and I am not impressed with the overall feature set of Github Actions. Running a CI matrix is tedious and the setup stage (installing dependencies, setting env variables, etc) has been plenty painful.

@ProfFan I would love to see how you set up all the different jobs for the CI.

ProfFan, 2020-03-21:
@varunagrawal https://github.com/ProfFan/gtsam/blob/feature/github_actions_ci/.github/workflows/build.yml

ProfFan, 2020-07-25:
I am bringing this issue out again that:
1. The CI is fully working with Github Actions
2. No timeouts with Github Actions
3. Windows builds on Github actions is running smoothly

So I propose to remove AppVeyor and Travis (in the PyBind branch).

@dellaert @jlblancoc @varunagrawal For comments.

dellaert, 2020-07-25:
sure :-)

jlblancoc, 2020-07-25:
I was thinking of proposing you CircleCi to avoid the time outs, but if you have actions already set up and running... brilliant! LGTM

varunagrawal, 2020-07-26:
Yes please! Whatever speeds up the CI pipeline (and you made excellent arguments for GitHub Actions). Minimizing the context switching should help with productivity immensely.

varunagrawal, 2020-09-22:
We got this working!! :tada:

closed 2020-09-22 by varunagrawal