borglab/gtsam #237: Mismatch between calibrate functions for Cal3_S2 and Cal3Bundler

varunagrawal, 2020-03-04:
## Description

`PinholeCamera` can accept both `Cal3_S2` and `Cal3Bundler` as calibrations, but the `calibrate` function is defined differently for each, such that when `camera.backproject(pn, depth)` is called, `Cal3Bundler` errors out.

## Steps to reproduce

This unit test when added to `testPinholeCamera.cpp` raises the error:

```cpp
TEST( PinholeCamera, Cal3Bundler) {
  Cal3Bundler calibration;
  Pose3 wTc;
  PinholeCamera<Cal3Bundler> camera(wTc, calibration);
  Point2 p(50, 100);
  Point3 point = camera.backproject(p, 10);
}
```

## Expected behavior

`PinholeCamera` (and primarily it's base class `PinholeBase`) should be able to call `calibration().calibrate` without any errors for all calibration types.

## Environment

N/A

## Additional information

varunagrawal, 2020-03-04:
This issue is easy to fix once we figure out how to handle the tolerance argument in `Cal3Bundler` so that the method signature is consistent.

johnwlambert, 2020-09-22:
@varunagrawal It looks like we can't access the `camera` parameter of `SfmData` in the wrapper until we resolve this (https://github.com/borglab/gtsam/blob/develop/gtsam/gtsam.i#L2865). Any idea for a path forward for resolving the tolerance argument issue?

varunagrawal, 2020-09-22:
This is something @dellaert and I need to discuss at a design level so I can't comment on that *sigh*.

I've had a PR ready for months actually, but it breaks the API signature which is what is causing the delay.

dellaert, 2020-09-22:
I’m sorry, what’s the issue?

varunagrawal, 2020-09-22:
`Cal3_S2` has `calibrate` defined as

```cpp
Point2 uncalibrate(const Point2& p, OptionalJacobian<2,5> Dcal = boost::none,
      OptionalJacobian<2,2> Dp = boost::none) const;
```

while `Cal3Bundler` has it defined as
```cpp
Point2 calibrate(const Point2& pi, const double tol = 1e-5) const;
```

aka no `Dcal` and `Dp`, so this causes issues with wrapping and use cases in optimization.

varunagrawal, 2020-09-22:
@dellaert the original PR I made is here: https://github.com/borglab/gtsam/pull/257

dellaert, 2020-09-22:
“Causes issues” ?

varunagrawal, 2020-09-22:
Using `CalBundler` gives the no-matching function error for `calibrate`. Actually, I think I can solve this in 15 minutes later tonight.

johnwlambert, 2020-09-22:
Thanks Varun. So Pybind11 will not allow two separate classes to have the same method with different signatures? That seems extremely restrictive. Or is it the case that these two classes are related in a particular hierarchy and that causes problems?

varunagrawal, 2020-09-23:
This was definitely an issue with the old Python wrapper so I don't know if this will work with Pybind11 or not.
My guess is that it won't because `PinholeCamera` has a `backproject` method that needs a `calibrate` with Jacobians, so if Pybind11 performs static analysis (via the compiler, which it should), it will see that code path as non-existential which is good.

On the other hand, even if you're able to wrap the function, if you try to perform a backprojection, your code will just throw an unhelpful segfault.

PR #539 (2020-09-29): Add calibrate with jacobians for Cal3Bundler
Added an additional `calibrate` method for `Cal3Bundler` which compute jacobians.

Since the normal calibration method is iterative, I have approximated the jacobian as if a single iteration step has occurred, which should suffice assuming the projection is close to the ideal measurements.

I would definitely like a thorough review on this though.

Fixes #237
closed 2020-09-29 by varunagrawal