borglab/gtsam #238: SymmetricBlockMatrix setzero problem.

shaolinbit, 2020-03-04:
In SymmetricBlockMatrix.h

/// Set the entire active matrix zero.
    void setZero() {
      full().triangularView<Eigen::Upper>().setZero();
    }

When matrix_.resize() is used in Hessian factor initialization, matrix_ may be not all zero since it may use a bulk of  dirty memory.
Maybe it should be written as follows.
   void setZero() {
      matrix() .setZero();
    }

varunagrawal, 2020-03-06:
I don't think this is a bug, because even with an increase in rows and columns for the symmetric matrix, we only care about the upper triangular part for elimination. As long as that is set to 0, we should be fine.

I'll let @dellaert and @richardroberts chime in.

varunagrawal, 2020-03-06:
Function in question is here https://github.com/borglab/gtsam/blob/develop/gtsam/base/SymmetricBlockMatrix.h#L263

shaolinbit, 2020-03-06:
[image]
[image]

Try this example. You will get the matrix results which is not zero after resize. Triangular set zero is no enough for some cases. 
/* ----------------------------------------------------------------------------

 * GTSAM Copyright 2010, Georgia Tech Research Corporation,
 * Atlanta, Georgia 30332-0415
 * All Rights Reserved
 * Authors: Frank Dellaert, et al. (see THANKS for the full author list)

 * See LICENSE for the license information

 * -------------------------------------------------------------------------- */

/**
 * @file RangeISAMExample_plaza1.cpp
 * @brief A 2D Range SLAM example
 * @date June 20, 2013
 * @author FRank Dellaert
 */

// Both relative poses and recovered trajectory poses will be stored as Pose2 objects
#include <gtsam/geometry/Pose2.h>

// Each variable in the system (poses and landmarks) must be identified with a unique key.
// We can either use simple integer keys (1, 2, 3, ...) or symbols (X1, X2, L1).
// Here we will use Symbols
#include <gtsam/inference/Symbol.h>

// We want to use iSAM2 to solve the range-SLAM problem incrementally
#include <gtsam/nonlinear/ISAM2.h>

// iSAM2 requires as input a set set of new factors to be added stored in a factor graph,
// and initial guesses for any new variables used in the added factors
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>

// We will use a non-liear solver to batch-inituialize from the first 150 frames
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>

// In GTSAM, measurement functions are represented as 'factors'. Several common factors
// have been provided with the library for solving robotics SLAM problems.
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/sam/RangeFactor.h>
#include <gtsam/slam/dataset.h>

// Standard headers, added last, so we know headers above work on their own
#include <fstream>
#include <iostream>

using namespace std;
using namespace gtsam;
namespace NM = gtsam::noiseModel;

// data available at http://www.frc.ri.cmu.edu/projects/emergencyresponse/RangeData/
// Datafile format (from http://www.frc.ri.cmu.edu/projects/emergencyresponse/RangeData/log.html)

// load the odometry
// DR: Odometry Input (delta distance traveled and delta heading change)
//    Time (sec)  Delta Dist. Trav. (m) Delta Heading (rad)
typedef pair<double, Pose2> TimedOdometry;
list<TimedOdometry> readOdometry() {
  list<TimedOdometry> odometryList;
  string data_file = findExampleDataFile("Plaza2_DR.txt");
  ifstream is(data_file.c_str());

  while (is) {
    double t, distance_traveled, delta_heading;
[... 121 lines cut ...]
        initial = batchOptimizer.optimize();
        gttoc_(batchInitialization);
        initialized = true;
      }
      gttic_(update);
      isam.update(newFactors, initial);
      gttoc_(update);
      gttic_(calculateEstimate);
      Values result = isam.calculateEstimate();
      gttoc_(calculateEstimate);
      lastPose = result.at<Pose2>(i);
      newFactors = NonlinearFactorGraph();
      initial = Values();
      countK = 0;
    }
    i += 1;
    //--------------------------------- odometry loop -----------------------------------------
  } // end for
  gttoc_(iSAM);

  // Print timings
  tictoc_print_();

  exit(0);
}

dellaert, 2020-03-06:
@shaolinbit it's a bit hard to see what you mean, sorry about that. How about adding a test in a fork in testSymmetricBlockMatrix.cpp that fails? You can do a PR with it or just paste a link to a 
 github gist with the test, and we'll look at it.

shaolinbit, 2020-03-06:
When you use an Eigen resize function, you may get a matrix which may not be zero. The resize function allocate a memory bulk for the matrix, this bulk is not guaranteed zero. Thus,  if you only set half a part of the matrix to zero, the results are not right too. After setting zero,Rand 1 update will update the whole matrix which is not zero.  

   info_.setZero();
  for(const auto& factor: factors)
    if (factor)
      factor->updateHessian(keys_, &info_);
  gttoc(update);

dellaert, 2020-03-06:
I know that you are worried about resize. But I don't see a resize in your code. I am looking for the specific gtsam use case that will yield a problem.

shaolinbit, 2020-03-06:
In this current edition, Hessianfactor.cpp, Line 252.
/* ************************************************************************* */
HessianFactor::HessianFactor(const GaussianFactorGraph& factors,
    const Scatter& scatter) {
  gttic(HessianFactor_MergeConstructor);

  Allocate(scatter);

  // Form A' * A
  gttic(update);
  info_.setZero();
  for(const auto& factor: factors)
    if (factor)
      factor->updateHessian(keys_, &info_);
  gttoc(update);
}

dellaert, 2020-03-06:
Great, that could be the problem. However, can you give me a *test* that fails? I.e., a unit test that calls this function (directly or indirectly) and exposes the problem?

closed 2020-03-07 by shaolinbit

dellaert, 2020-03-07:
@shaolinbit What's the reason for closing? Did you realize it was not an issue?

shaolinbit, 2020-03-08:
> @shaolinbit What's the reason for closing? Did you realize it was not an issue?

Sorry, U are right. This is not an issue. 
However, I tested the above example. It would collapse at CholeskyPartial. It is difficult for me to seek the origin of this bug. 
[image]

dellaert, 2020-03-08:
Thanks! Seems like you’re deep inside GTSAM. That’s non-obvious code :-)
If there is a bug you are encountering with GTSAM the best advice I can give you is to create a minimal failing unit test that calls GTSAM through the normal API, and share it with us.

shaolinbit, 2020-03-08:
> Thanks! Seems like you’re deep inside GTSAM. That’s non-obvious code :-)
> If there is a bug you are encountering with GTSAM the best advice I can give you is to create a minimal failing unit test that calls GTSAM through the normal API, and share it with us.

I have spent two years in gtsam. Now I am trying to code a cuda version. I am not very familiar with github. 
Of course, I will report the bug if I encounter one.

dellaert, 2020-03-08:
Awesome ! @dongjing3309 and @lvzhaoyang actually tried CUDA as a solver in MiniSAM, and it did not do a great job, performance-wise, if I recall correctly. It will require some thought :-) Easiest way is actually to replace the entire linear solver back-end, something @ProfFan and I are working to make possible (e.g, use SuiteSparse). If you'd like to chat/be involved, send me an email at frank@cc.gatech.edu and we can go from there.