borglab/gtsam #240: TBB is not propagated by CMake

yrzhanov, 2020-03-06:
I use CMake-GUI (3.17.0) on Windows 10 (VS 2015) and set locations of tbb.lib and tbbmalloc.lib.
Nevertheless, the generated  *.vcxproj files for examples and timing contain TBB-NOTFOUND

varunagrawal, 2020-03-29:
@jlblancoc please can you take a look at this? I don't have a windows machine to test this one and you are the CMake expert here. I can assist with TBB related questions and support if you need any.

jorigin, 2020-06-02:
Hello,

I'm actually trying to compile GTSAM:
OS: Windows 10 x64
TBB: 2020.2
BOOST: 1.70.0 dynamic
MSVC: 2019

and with the standard behavior all the generated *.vcxproj files contains reference to tbb-NOT-FOUND

I've been able to avoid this by the following actions:

* Replace gtsam\CMakeLists.txt line
    set(Boost_USE_STATIC_LIBS ON)
by
    set(Boost_USE_STATIC_LIBS OFF)
 
* Replace gtsam\cmake\FindTBB.cmake line
    add_library(${libname} SHARED IMPORTED)"
by
    add_library(${libname} STATIC IMPORTED)"

The compilation is working after these changes.

yrzhanov, 2020-06-03:
Hi,

I’m using Windows 10 x64, VS2015, boost 1.68.0

TBB propagated fine now, and libraries compile ok, but examples behave strangely. Although I’m linking as :

 

..\lib\Debug\libgtsamDebug.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_program_options-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_serialization-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_filesystem-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_thread-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_atomic-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_date_time-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_regex-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_timer-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_chrono-vc140-mt-gd-x64-1_68.lib;C:\boost_1_68_0\lib64-msvc-14.0\boost_system-vc140-mt-gd-x64-1_68.lib;C:\Program Files (x86)\IntelSWTools\compilers_and_libraries_2019.1.144\windows\tbb\lib\intel64_win\vc14\tbb_debug.lib;C:\Program Files  (x86)\IntelSWTools\compilers_and_libraries_2019.1.144\windows\tbb\lib\intel64_win\vc14\tbbmalloc_debug.lib;..\lib\Debug\metisDebug.lib;kernel32.lib;user32.lib;gdi32.lib;winspool.lib;shell32.lib;ole32.lib;oleaut32.lib;uuid.lib;comdlg32.lib;advapi32.lib

 

Linking fails with:

 

1>------ Build started: Project: CameraResectioning, Configuration: Debug x64 ------

1>LINK : fatal error LNK1104: cannot open file 'libboost_serialization-vc140-mt-gd-x64-1_68.lib'

========== Build: 0 succeeded, 1 failed, 3 up-to-date, 0 skipped ==========

 

Don’t understand where “libboost” comes from. Could you help, please?

 

Thanks,

    Yuri

 

===============================================================================

Yuri Rzhanov

Center for Coastal and Ocean Mapping/Joint Hydrographic Center

University of New Hampshire

Tel: 1-603-8620075

 

Let go all your heartaches, it's as easy as being free 
You don't have to wait a while

varunagrawal, 2020-12-01:
@yrzhanov this seems to be an issue with your boost installation. Please ensure you have the seralization packages of boost installed as well. If you're still having issues, please ask questions on the Google Group or feel free to reopen this issue.

closed 2020-12-01 by varunagrawal

whu-lyh, 2021-06-22:
> Hello,
>
> I'm actually trying to compile GTSAM:
> OS: Windows 10 x64
> TBB: 2020.2
> BOOST: 1.70.0 dynamic
> MSVC: 2019
>
> and with the standard behavior all the generated *.vcxproj files contains reference to [...]

Thank you, this do works.
BTW,  i still wander this solution is official solution that i have tbb as shared libraries and boost is complied as static library as default. Why should i do this "add_library(${libname} STATIC IMPORTED)" operation seeming funny to me :)