borglab/gtsam #244: Improved numerical stability for comparison of matrices and vectors

varunagrawal, 2020-03-08:
## Feature

Improve the numerical stability of comparison for matrices and vectors.
Reference: https://floating-point-gui.de/errors/comparison/

## Motivation

Currently, the comparison operation follows an older floating point comparison method which is unstable when the matrix/vector has both small and large values together. The larger values lose precision because of the smaller values and thus the comparison fails.

This is especially important since tests on calibration matrices with radial distortion can fail due to 
the instability.

## Pitch

Follow the correct floating point comparison method.

## Alternatives

It's an IEEE 754 standard, so no other option really.

## Additional context

varunagrawal, 2020-03-08:
@dellaert @ProfFan I have a working implementation of this. Would you two have time to review this weekend?

ProfFan, 2020-03-08:
I think I will give a shot.

varunagrawal, 2020-03-17:
Fixed by #245

closed 2020-03-17 by varunagrawal