borglab/gtsam #253: Tests fail when using quaternion representation

varunagrawal, 2020-03-18:
## Description

Various tests fail when using the Quaternion representation of rotations. These same tests pass when using rotation matrices.

Specific test files are:

1. testImuFactor
2. testNavState
3. testScenario

## Steps to reproduce

1. Compile GTSAM with `GTSAM_USE_QUATERNIONS=ON` cmake flag.
2. Run tests.

## Expected behavior

All tests should pass indicating behavior similar to Rotation Matrices.

## Environment

Arch Linux with GCC 9.2.0.

dellaert, 2020-03-18:
Is there not a Travis confit that checks the quarter ion setting? There should be, maybe add in “special” section.

varunagrawal, 2020-04-08:
Linking an answer from StackOverflow that explains the issue really well

https://stackoverflow.com/questions/60749047/correct-way-to-use-eigentransposeconst-matrix3/60937475#60937475

ProfFan, 2020-04-09:
IMHO this should be solved with either:
1. Just copy it.
2. Just copy it.

as we are calculating this value anyway :)

dellaert, 2020-04-09:
Yes, let's just return a Matrix3 ? Is that not what is happening now? @varunagrawal close if you agree.

varunagrawal, 2020-05-19:
Closing because this involves an unnecessary copy operation.

closed 2020-05-19 by varunagrawal