borglab/gtsam #261: The difference between the theoretical derivation of ImuFactor and its implementation in CombinedImuFactor.cpp

LiShuaixin, 2020-03-26:
Hi, I found that you've marked an 'TODO' at the line 81 in `[CombinedImuFactor.cpp](https://github.com/borglab/gtsam/blob/develop/gtsam/navigation/CombinedImuFactor.cpp)` and noted the existing issue. 

For the matrix `F` in the code, the block representing the derivative of position with respect to acceleration should not be set zero theoretically. I think it should be the second part of `B`, i.e. `B->block<3, 3>(3, 0)`, which obviously not equal to zero as what have been done in `UpdatePreintegrated()`.   

I saw your mark, and wandered if there is any other reason for the current implementation?

dellaert, 2020-04-04:
It's been a while since I looked at this code. Can you think of a unit test that would fail now and succeed if we fix it?

varunagrawal, 2023-06-15:
I think I fixed this when I refactored the CombinedImuFactor.

closed 2023-06-15 by varunagrawal