borglab/gtsam #262: IMUFactor example

kvmanohar22, 2020-03-29:
Hi,

I am going through the examples and for IMUFactor example given in [ImuFactorsExample](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample.cpp), although the integration is perfomed for each new IMU measurement, the integrated measurement is added to the graph for each new IMU measurement.

But as proposed in [1] and from my understanding of the theory, the advantage lies in integrating over several measurements and then adding that integrated measurement to the graph.

So, if it is okay, I would love to add an example which follows the intent proposed in [1] using isam2.

Let me know what your thoughts are on this.

> [1] Christian Forster, Luca Carlone, Frank Dellaert, and Davide Scaramuzza, "IMU Preintegration on Manifold for Efficient Visual-Inertial Maximum-a-Posteriori Estimation", Robotics: Science and Systems (RSS), 2015.

dellaert, 2020-03-29:
Integration has to be done for each measurement. But the preintegrated factor is added to the graph at lower frequency.

mikesheffler, 2020-03-29:
Yes. Note that `graph->add(imu_factor);` [only happens inside](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample.cpp#L201) of the `else if (type == 1) { // GPS measurement` block 😃

The example does use `LevenbergMarquardtOptimizer`, however. If you'd like to create an option to use `ISAM2`, that would be great. The [IMUFactorsExample2](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorExample2.cpp) and [MATLAB IMUKittiExampleGPS](https://github.com/borglab/gtsam/blob/develop/matlab/gtsam_examples/IMUKittiExampleGPS.m)  examples are similar and already use `ISAM2`; I'm sure it would work just fine here as an option, if it doesn't make the code too convoluted.

kvmanohar22, 2020-03-30:
Yes, I think adding as an option would be great.

varunagrawal, 2020-09-22:
Closing this since the #263 is merged.

closed 2020-09-22 by varunagrawal