borglab/gtsam #265: Automatic Differentiation issue when using Expressions with Dynamic Matrices

varunagrawal, 2020-04-01:
## Description

If we use an expression with dynamic matrices, it fails to compile.

Here is a unit test that doesn't pass:

```cpp
Eigen::Matrix<double, -1, 1> vectorSquare(Eigen::Matrix<double, -1, 1> x,
    OptionalJacobian<-1, -1> H = boost::none) {
    if (H) *H = (2 * x.asDiagonal());
  return x.cwiseProduct(x);
}

TEST(Expression, DynamicMatrix) {
  const size_t M = 12;
  Expression<Eigen::Matrix<double, -1, 1>> V_(symbol('V', 0));
  Expression<Eigen::Matrix<double, -1, 1>> vecSq_(vectorSquare, V_);

  Values values;
  values.insert<Eigen::Matrix<double, -1, 1>>(symbol('V', 0), Eigen::Matrix<double, -1, 1>::Ones(M, 1)*3);

  // std::vector<Eigen::Matrix<double, -1, 1>> H(1);
  Eigen::Matrix<double, -1, 1> expected = Eigen::Matrix<double, -1, 1>::Ones(M, 1)*9;
  Eigen::Matrix<double, -1, 1> actual = vecSq_.value(values);
  EXPECT(assert_equal(expected, actual, 1e-8));
  // EXPECT_CORRECT_EXPRESSION_JACOBIANS(vecSq_, values, 1e-5, 1e-9);
}
```

Please see the stack trace at the end of the issue.

I believe this is (partially) due to the way the leaf node jacobians are initialized in `ExecutionTrace.h`.

https://github.com/borglab/gtsam/blob/e2ed25328b2ff75638301ed466bd9921691d4c90/gtsam/nonlinear/internal/ExecutionTrace.h#L151

Indeed, the error can be reproduced with a simple line like 
```cpp
Eigen::Matrix<double, -1, -1> I = Eigen::Matrix<double, -1, -1>::Identity();
```

## Steps to reproduce

1. Add above unit test to `testExpression.cpp`.
2. Run the unit test.

## Expected behavior

The expression should work just like a version with fixed size matrices.

## Environment

OS: Arch Linux
Branch: develop

## Additional information

```sh
In file included from /home/varun/borglab/gtsam/gtsam/nonlinear/internal/ExpressionNode.h:23,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/Expression-inl.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/Expression.h:302,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/ExpressionFactor.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/expressionTesting.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/tests/testExpression.cpp:20:
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/CallRecord.h: In instantiation of ‘struct gtsam::internal::CallRecord<-1>’:
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/CallRecord.h:140:8:   required from ‘struct gtsam::internal::CallRecordImplementor<gtsam::internal::UnaryExpression<Eigen::Matrix<double, -1, 1>, Eigen::Matrix<double, -1, 1> >::Record, -1>’
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/ExpressionNode.h:272:10:   required from ‘struct gtsam::internal::UnaryExpression<Eigen::Matrix<double, -1, 1>, Eigen::Matrix<double, -1, 1> >::Record’
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/ExpressionNode.h:239:34:   required from ‘gtsam::internal::UnaryExpression<T, A1>::UnaryExpression(gtsam::internal::UnaryExpression<T, A1>::Function, const gtsam::Expression<A>&) [with T = Eigen::Matrix<double, -1, 1>; A1 = Eigen::Matrix<double, -1, 1>; gtsam::internal::UnaryExpression<T, A1>::Function = boost::function<Eigen::Matrix<double, -1, 1>(const Eigen::Matrix<double, -1, 1>&, gtsam::OptionalJacobian<-1, -1>)>]’
/home/varun/borglab/gtsam/gtsam/nonlinear/Expression-inl.h:55:11:   required from ‘gtsam::Expression<T>::Expression(typename gtsam::Expression<T>::UnaryFunction<A>::type, const gtsam::Expression<A>&) [with A = Eigen::Matrix<double, -1, 1>; T = Eigen::Matrix<double, -1, 1>; typename gtsam::Expression<T>::UnaryFunction<A>::type = boost::function<Eigen::Matrix<double, -1, 1>(const Eigen::Matrix<double, -1, 1>&, gtsam::OptionalJacobian<-1, -1>)>]’
/home/varun/borglab/gtsam/gtsam/nonlinear/tests/testExpression.cpp:543:67:   required from here
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/CallRecord.h:112:16: error: ‘void gtsam::internal::CallRecord<T>::_reverseAD3(const Eigen::Matrix<double, -1, Cols>&, gtsam::internal::JacobianMap&) const [with int Cols = -1]’ cannot be overloaded with ‘void gtsam::internal::CallRecord<T>::_reverseAD3(const Matrix&, gtsam::internal::JacobianMap&) const [with int Cols = -1; gtsam::Matrix = Eigen::Matrix<double, -1, -1>]’
[... 30 lines cut ...]
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/internal/JacobianMap.h:23,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/Expression.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/ExpressionFactor.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/expressionTesting.h:22,
                 from /home/varun/borglab/gtsam/gtsam/nonlinear/tests/testExpression.cpp:20:
/home/varun/borglab/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/CwiseNullaryOp.h: In instantiation of ‘static const IdentityReturnType Eigen::MatrixBase<Derived>::Identity() [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::MatrixBase<Derived>::IdentityReturnType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<double>, Eigen::Matrix<double, -1, -1> >; typename Eigen::DenseBase<Derived>::PlainObject = Eigen::Matrix<double, -1, -1>; typename Eigen::internal::traits<T>::Scalar = double]’:
/home/varun/borglab/gtsam/gtsam/nonlinear/internal/ExecutionTrace.h:151:55:   required from ‘void gtsam::internal::ExecutionTrace<T>::startReverseAD1(gtsam::internal::JacobianMap&) const [with T = Eigen::Matrix<double, -1, 1>]’
/home/varun/borglab/gtsam/gtsam/nonlinear/Expression-inl.h:210:3:   required from ‘T gtsam::Expression<T>::valueAndJacobianMap(const gtsam::Values&, gtsam::internal::JacobianMap&) const [with T = Eigen::Matrix<double, -1, 1>]’
/home/varun/borglab/gtsam/gtsam/nonlinear/Expression-inl.h:173:14:   required from ‘T gtsam::Expression<T>::valueAndDerivatives(const gtsam::Values&, const KeyVector&, gtsam::FastVector<int>&, std::vector<Eigen::Matrix<double, -1, -1> >&) const [with T = Eigen::Matrix<double, -1, 1>; gtsam::KeyVector = std::vector<long unsigned int, std::allocator<long unsigned int> >; gtsam::FastVector<int> = std::vector<int, std::allocator<int> >]’
/home/varun/borglab/gtsam/gtsam/nonlinear/Expression-inl.h:140:12:   required from ‘T gtsam::Expression<T>::value(const gtsam::Values&, boost::optional<std::vector<Eigen::Matrix<double, -1, -1> >&>) const [with T = Eigen::Matrix<double, -1, 1>]’
/home/varun/borglab/gtsam/gtsam/nonlinear/tests/testExpression.cpp:551:60:   required from here
/home/varun/borglab/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/StaticAssert.h:145:46: error: static assertion failed: YOU_CALLED_A_FIXED_SIZE_METHOD_ON_A_DYNAMIC_SIZE_MATRIX_OR_VECTOR
  145 |   EIGEN_STATIC_ASSERT(TYPE::SizeAtCompileTime!=Eigen::Dynamic, \
      |                             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/varun/borglab/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/StaticAssert.h:33:54: note: in definition of macro ‘EIGEN_STATIC_ASSERT’
   33 |     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
      |                                                      ^
/home/varun/borglab/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/CwiseNullaryOp.h:702:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT_FIXED_SIZE’
  702 |   EIGEN_STATIC_ASSERT_FIXED_SIZE(Derived)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [gtsam/nonlinear/tests/CMakeFiles/testExpression.dir/build.make:63: gtsam/nonlinear/tests/CMakeFiles/testExpression.dir/testExpression.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:11995: gtsam/nonlinear/tests/CMakeFiles/testExpression.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:12504: gtsam/nonlinear/tests/CMakeFiles/testExpression.run.dir/rule] Error 2
make: *** [Makefile:4617: testExpression.run] Error 2
```

dellaert, 2020-04-01:
Doh! This might be the real reason DOubleExpresso uses fixed degree.
This is maybe not a bug, but an (unfortunate) design choice. Let’s talk.

dellaert, 2020-08-02:
By design, not a bug

closed 2020-08-02 by dellaert