borglab/gtsam #266: Compilation error Ubuntu 16.04: "4" "GREATER_EQUAL" "2020"

ToniRV, 2020-04-01:
## Description

Our CI building Ubuntu 16.04 and installing GTSAM started failing recently (it works fine for Ubuntu 18.04).
It seems like an update in the CMakeLists.txt might have broken something.
Here is the error:
```
  if given arguments:

    "4" "GREATER_EQUAL" "2020"

  Unknown arguments specified
```

## command:

```
cd gtsam 
git fetch 
mkdir build 
cd build 
cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON .. 
make -j$(nproc) install
```

## full error:

```
Step 18/27 : RUN cd gtsam &&     git fetch &&     mkdir build &&     cd build &&     cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON .. &&     make -j$(nproc) install
 ---> Running in b64663cad0af
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/root/gtsam]
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.43") found components:  serialization system filesystem thread program_options date_time timer chrono regex atomic 
-- Found TBB: /usr/include (found suitable version "4.4", minimum required is "4.4") found components:  tbb tbbmalloc 
[91mCMake Error at CMakeLists.txt:203 (if):
  if given arguments:

    "4" "GREATER_EQUAL" "2020"

  Unknown arguments specified

[0m-- Configuring incomplete, errors occurred!
See also "/root/gtsam/build/CMakeFiles/CMakeOutput.log".
See also "/root/gtsam/build/CMakeFiles/CMakeError.log".
The command '/bin/sh -c cd gtsam &&     git fetch &&     mkdir build &&     cd build &&     cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON .. &&     make -j$(nproc) install' returned a non-zero code: 1
```
Is there any backwards compatibility break we might need to be aware of?
Perhaps it's something from a dependency that breaks?

varunagrawal, 2020-04-01:
We updated the `CMakeLists.txt` recently to account for differences in versions of TBB 4.4 and 2020.

Let me try and reproduce this error. I may have a quick fix for it.

ToniRV, 2020-04-01:
Awesome thanks Varun!

varunagrawal, 2020-04-01:
I know what the issue is, I ran into it myself a few days ago when I was working on the TBB PR, but it took me a while to remember it.

What is your cmake version? I would believe it is less than `3.15`? If so, add these lines to your Dockerfile for your CI system

```docker
#install latest cmake
ADD https://cmake.org/files/v3.15/cmake-3.15.7-Linux-x86_64.sh /cmake-3.15.7-Linux-x86_64.sh
RUN mkdir /opt/cmake
RUN sh /cmake-3.15.7-Linux-x86_64.sh --prefix=/opt/cmake --skip-license
RUN ln -s /opt/cmake/bin/cmake /usr/local/bin/cmake
```

ToniRV, 2020-04-01:
Ok, trying right now, thank you.
Do you think we could remove this requirement though? IMHO, it seems quite intrusive to ask users to update cmake...

varunagrawal, 2020-04-01:
That's a fair ask, though I am not sure what is the equivalent command in lower versions of CMake. Any clues?

varunagrawal, 2020-04-01:
Also a counter-argument could be that Ubuntu 16.04 is nearing the end of its lifecycle in a year so upgrading your systems would be a good idea.

PR #268 (2020-04-04): Maintain backwards compatibility of CMake
Updated the TBB version check so it is now backwards compatible.
Should fix #266

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/268)
closed 2020-04-04 by dellaert