borglab/gtsam #278: Bild error

emma010165, 2020-04-09:
Hi everyone !

I got the following build error:

Can anyone help me solve this problem.
Thanks.

varunagrawal, 2020-04-09:
What version of boost are you using?

emma010165, 2020-04-09:
@varunagrawal I am using Boost version 1.65

ProfFan, 2020-04-13:
@emma010165 Please attach all the relevant logs and the commands you have used to build GTSAM.

emma010165, 2020-04-22:
Hi, I have changed my gcc version from 4.8 to a much newer one gcc 7 and it worked perfectly (same commands in readme for building GTSAM). Even though gcc4.8 is said to be supported for building gtsam, I don't really know why it didn't work for me. Thank you for your responses @profFan and @varunagrawal. I guess I'll close this issue.

closed 2020-04-22 by emma010165