borglab/gtsam #282: Error using gtsam toolbox, Error using gtsam_wrapper

hunkyu, 2020-04-28:
## Description
>> test_gtsam
Starting: testCal3Unified
Starting: testKalmanFilter
Starting: testJacobianFactor
Starting: testValues
Starting: testPriorFactor
Starting: testLocalizationExample
Starting: testOdometryExample
Starting: testPlanarSLAMExample
Starting: testPose2SLAMExample
Starting: testPose3SLAMExample
Starting: testSFMExample
Error using gtsam_wrapper
Exception from gtsam:
wrap internal error: attempted wrap of invalid type

Error in gtsam.PinholeCameraCal3_S2.Lookat (line 202)
        gtsam_wrapper(3071, varargin{:});

Error in gtsam.VisualISAMGenerateData (line 34)
    truth.cameras{i} = PinholeCameraCal3_S2.Lookat(t, Point3, Point3([0,0,1]'), truth.K);

Error in testSFMExample (line 19)
[data,truth] = VisualISAMGenerateData(options);

Error in test_gtsam (line 39)
testSFMExample
 
>> gtsam_wrapper

--------------------------------------------------------------------------------
       Segmentation violation detected at 二 4月 28 15:01:11 2020 +0800
--------------------------------------------------------------------------------

Configuration:
  Crash Decoding           : Disabled - No sandbox or build area path
  Crash Mode               : continue (default)
  Default Encoding         : UTF-8
  Deployed                 : false
  Desktop Environment      : ubuntu:GNOME
  GNU C Library            : 2.27 stable
  Graphics Driver          : Unknown hardware 
  Java Version             : Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  MATLAB Architecture      : glnxa64
  MATLAB Entitlement ID    : 6257193
  MATLAB Root              : /usr/local/MATLAB/R2018b
  MATLAB Version           : 9.5.0.944444 (R2018b)
  OpenGL                   : hardware
  Operating System         : Ubuntu 18.04.4 LTS
  Process ID               : 7805
  Processor ID             : x86 Family 6 Model 142 Stepping 12, GenuineIntel
  Session Key              : 86e71aa7-907c-4da4-aa74-74a6a031b243
  Static TLS mitigation    : Enabled: Full
  Window System            : The X.Org Foundation (12005000), display :0

Fault Count: 1

Abnormal termination

Register State (from fault):
  RAX = 0000000000000000  RBX = 0000000000000000
  RCX = 00007f192a14d6e0  RDX = 00007f191f249030
  RSP = 00007f1a01eadaa0  RBP = 00007f1a01eadab0
  RSI = 00007f18967d8054  RDI = 0000000000000000

   R8 = 0000000000000068   R9 = 00007f1a1c260476
  R10 = 0000000000000000  R11 = 00007f1a1c260192
  R12 = 00007f18967d8054  R13 = 0000000000000000
  R14 = 00007f1a01eade40  R15 = 00007f1a01eab690

  RIP = 00007f1a241a352c  EFL = 0000000000010206

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f1a241a352c                               bin/glnxa64/libmx.so+00521516 _ZN6matrix6detail10noninlined12mx_array_api6mxGetMEPK11mxArray_tag+00000012
[  1] 0x00007f189644b0cf          /home/hunkyu/toolbox/gtsam_wrapper.mexa64+06668495 _Z11checkScalarPK11mxArray_tagPKc+00000015
[  2] 0x00007f189644b5d0          /home/hunkyu/toolbox/gtsam_wrapper.mexa64+06669776 _Z6unwrapIiET_PK11mxArray_tag+00000016
[  3] 0x00007f18965d4133          /home/hunkyu/toolbox/gtsam_wrapper.mexa64+08278323 mexFunction+00000179
[  4] 0x00007f1a10143f80                              bin/glnxa64/libmex.so+00491392
[  5] 0x00007f1a10144347                              bin/glnxa64/libmex.so+00492359
[  6] 0x00007f1a10144f93                              bin/glnxa64/libmex.so+00495507
[  7] 0x00007f1a1013085c                              bin/glnxa64/libmex.so+00411740
[  8] 0x00007f1a11ab1bc8                   bin/glnxa64/libmwm_dispatcher.so+00973768 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00001240
[  9] 0x00007f1a11ab202e                   bin/glnxa64/libmwm_dispatcher.so+00974894 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000030
[ 10] 0x00007f1a0de34297                          bin/glnxa64/libmwm_lxe.so+12862103
[ 11] 0x00007f1a0de34cff                          bin/glnxa64/libmwm_lxe.so+12864767
[ 12] 0x00007f1a0df3a219                          bin/glnxa64/libmwm_lxe.so+13935129
[ 13] 0x00007f1a0debf861                          bin/glnxa64/libmwm_lxe.so+13432929
[ 14] 0x00007f1a0d6a2b58                          bin/glnxa64/libmwm_lxe.so+04926296
[ 15] 0x00007f1a0d6a533c                          bin/glnxa64/libmwm_lxe.so+04936508
[ 16] 0x00007f1a0d6a13cd                          bin/glnxa64/libmwm_lxe.so+04920269
[ 17] 0x00007f1a0d69b0d1                          bin/glnxa64/libmwm_lxe.so+04894929
[ 18] 0x00007f1a0d69b309                          bin/glnxa64/libmwm_lxe.so+04895497
[ 19] 0x00007f1a0d6a0bf6                          bin/glnxa64/libmwm_lxe.so+04918262
[ 20] 0x00007f1a0d6a0cdf                          bin/glnxa64/libmwm_lxe.so+04918495
[ 21] 0x00007f1a0d7d97e9                          bin/glnxa64/libmwm_lxe.so+06199273
[ 22] 0x00007f1a0d7dcf03                          bin/glnxa64/libmwm_lxe.so+06213379
[ 23] 0x00007f1a0dd03b31                          bin/glnxa64/libmwm_lxe.so+11615025
[ 24] 0x00007f1a0dcb26ec                          bin/glnxa64/libmwm_lxe.so+11282156
[ 25] 0x00007f1a0dcb9818                          bin/glnxa64/libmwm_lxe.so+11311128
[ 26] 0x00007f1a0dcbb162                          bin/glnxa64/libmwm_lxe.so+11317602
[ 27] 0x00007f1a0dd49b1f                          bin/glnxa64/libmwm_lxe.so+11901727
[ 28] 0x00007f1a0dd49e0a                          bin/glnxa64/libmwm_lxe.so+11902474
[ 29] 0x00007f1a103b2122                         bin/glnxa64/libmwbridge.so+00221474 _Z8mnParserv+00000626
[ 30] 0x00007f1a11fb0c44                            bin/glnxa64/libmwmcr.so+00699460
[ 31] 0x00007f1a2518c7df                             bin/glnxa64/libmwms.so+02627551 _ZNSt13__future_base13_State_baseV29_M_do_setEPSt8functionIFSt10unique_ptrINS_12_Result_baseENS3_8_DeleterEEvEEPb+00000031
[ 32] 0x00007f1a2543a827              /lib/x86_64-linux-gnu/libpthread.so.0+00063527
[ 33] 0x00007f1a23d2e016                         bin/glnxa64/libmwmlutil.so+06909974 _ZSt9call_onceIMNSt13__future_base13_State_baseV2EFvPSt8functionIFSt10unique_ptrINS0_12_Result_baseENS4_8_DeleterEEvEEPbEJPS1_S9_SA_EEvRSt9once_flagOT_DpOT0_+00000102
[ 34] 0x00007f1a11fb0b23                            bin/glnxa64/libmwmcr.so+00699171
[ 35] 0x00007f1a2833527b                            bin/glnxa64/libmwmvm.so+03502715 _ZN14cmddistributor15PackagedTaskIIP10invokeFuncIN7mwboost8functionIFvvEEEEENS2_10shared_ptrINS2_13unique_futureIDTclfp_EEEEEERKT_+00000059
[ 36] 0x00007f1a28335368                            bin/glnxa64/libmwmvm.so+03502952 _ZNSt17_Function_handlerIFN7mwboost3anyEvEZN14cmddistributor15PackagedTaskIIP10createFuncINS0_8functionIFvvEEEEESt8functionIS2_ET_EUlvE_E9_M_invokeERKSt9_Any_data+00000024
[ 37] 0x00007f1a1234d08c                            bin/glnxa64/libmwiqm.so+00897164 _ZN7mwboost6detail8function21function_obj_invoker0ISt8functionIFNS_3anyEvEES4_E6invokeERNS1_15function_bufferE+00000028
[ 38] 0x00007f1a1234cd45                            bin/glnxa64/libmwiqm.so+00896325 _ZN3iqm18PackagedTaskPlugin7executeEP15inWorkSpace_tag+00000437
[ 39] 0x00007f1a1232b517                            bin/glnxa64/libmwiqm.so+00759063
[ 40] 0x00007f1a1230f6e8                            bin/glnxa64/libmwiqm.so+00644840
[ 41] 0x00007f1a1231020f                            bin/glnxa64/libmwiqm.so+00647695
[ 42] 0x00007f1a12310898                            bin/glnxa64/libmwiqm.so+00649368
[ 43] 0x00007f1a27d17da6                       bin/glnxa64/libmwservices.so+02993574
[ 44] 0x00007f1a27d19422                       bin/glnxa64/libmwservices.so+02999330
[ 45] 0x00007f1a27d19d3b                       bin/glnxa64/libmwservices.so+03001659 _Z25svWS_ProcessPendingEventsiib+00000187
[ 46] 0x00007f1a11fb332c                            bin/glnxa64/libmwmcr.so+00709420
[ 47] 0x00007f1a11fb3b08                            bin/glnxa64/libmwmcr.so+00711432
[ 48] 0x00007f1a11fabd92                            bin/glnxa64/libmwmcr.so+00679314
[ 49] 0x00007f1a254326db              /lib/x86_64-linux-gnu/libpthread.so.0+00030427
[ 50] 0x00007f1a26db088f                    /lib/x86_64-linux-gnu/libc.so.6+01185935 clone+00000063
[ 51] 0x0000000000000000                                   <unknown-module>+00000000

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.
** This crash report has been saved to disk as /home/hunkyu/matlab_crash_dump.7805-1 **

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment
ubuntu 18.04 
matlab R2018b
gtsam branch is develop

## Additional information

hunkyu, 2020-04-28:
I have solved it, just use the master branch

closed 2020-04-28 by hunkyu

phdsenwang, 2023-05-07:
Hello, I also encountered the same problem. Can you tell me your solution in detail? Thank you so much.