borglab/gtsam #286: Handling edgelet features

kvmanohar22, 2020-05-02:
## Feature
Current implementation of bundle adjustment is limited to corner features. I am proposing for handling additional type of features i.e, edgelets.

## Motivation
Tracking pixels having any gradient as in semi-dense and dense approaches could make tracking more reliable in environments having little texture.

## Pitch
Implementing bundle adjustment for edgelet features could be a good start. This is proposed in [1], I am reposting the optimization function here
[image]

The subtle change is to restrict reprojection error of edge features along the edge normal.

## Additional context
The usage of edgelet features is proposed in the following paper.

## References
[1] SVO: Semi-Direct Visual Odometry for Monocular and Multi-Camera Systems

dellaert, 2020-05-02:
@akshay-krishnan

dellaert, 2020-08-02:
Go for it ;-)

dellaert, 2020-08-02:
Note, you can add a factor, but image processing and other front-end code is deliberately not included with GTSAM - which is focused on the back-end optimization. There are several line-based factors that might already be similar to what you want.

varunagrawal, 2023-12-30:
@kvmanohar22 if you're still interested, we would love to have a PR for this feature. I am closing this issue for now since it is quite old.

closed 2023-12-30 by varunagrawal