borglab/gtsam #287: Problem with Python wrapper: No module named 'gtsam.utils'

Study-is-happy, 2020-05-03:
First I turn on the GTSAM_INSTALL_CYTHON_TOOLBOX flag and turn off the GTSAM_ALLOW_DEPRECATED_SINCE_V4 flag in CMakeList.txt in gtsam home directory

Here is what I did:
cd gtsam
mkdir build
cd build
cmake ..
sudo make install
cd cython
sudo python setup.py install

But I failed to run any python examples, with error No module named 'gtsam.utils'
I can import gtsam but I cannot import any sub packages of gtsam

I am in virtualenv (actived) and Ubuntu 18.04

Thank you for providing and maintaining such an amazing library for SLAM!!!

ProfFan, 2020-05-03:
Actually I think I encountered this problem when debugging in a Ubuntu 18.04 container. The simple solution is to go to the `build/cython/gtsam` folder, `ln -s $(pwd)/../../cython/gtsam/utils ./utils`, and run setup again.

ProfFan, 2020-05-03:
Short explanation: it failed to create the symlink to `utils`, so you can manually create it.

Study-is-happy, 2020-05-03:
> Short explanation: it failed to create the symlink to `utils`, so you can manually create it.

I linked utils, and I can see the red symlink in gtsam/build/cython/gtsam
[image]

And I run setup again, still not working :(

ProfFan, 2020-05-03:
Try `ls utils` and see if it's a broken symlink? Don't copy my command, the linked path should be in the source folder, a.k.a `<gtsam>/cython/gtsam/utils`.

Study-is-happy, 2020-05-03:
> Try `ls utils` and see if it's a broken symlink? Don't copy my command, the linked path should be in the source folder, a.k.a `<gtsam>/cython/gtsam/utils`.

I correct the link path and the symlink is working now, it turned blue. But I setup again and still not working :(

ProfFan, 2020-05-03:
@Study-is-happy do a cleanup by `pip uninstall gtsam` and do it again, if that still fails you can manually copy the contents of `utils` to the `site-packages/gtsam***` directory.

dellaert, 2020-05-03:
Try it in a new conda environment, making sure to run install with the environment’s python.

varunagrawal, 2020-07-30:
@Study-is-happy any updates on this?

We added a new `make python-install` command that should install everything for you (also cleaned up the cmake quite a bit).
Hopefully that works for you?

ProfFan, 2020-07-30:
I kind of know what is the problem, he is probably importing the gtsam "folder" because he is running the examples under some folder with a subfolder called `gtsam`.

dellaert, 2020-08-02:
Re-open if still an issue

closed 2020-08-02 by dellaert