borglab/gtsam #290: How to convert KITTI raw data to KittiGps_converted.txt ?

52stj, 2020-05-06:
I notice that the example data KittiGps_converted.txt is a kind of conversion of the raw kitti data of 2011_10_03_drive_0027 (17.6 GB) http://www.cvlibs.net/datasets/kitti/raw_data.php?type=residential.    

I have two questions:  

1. The coordinate in KittiGps_converted.txt is not start from the 0 timestamp position as the top two lines are:  
Time,X,Y,Z
46534.478375790000428,-6.8269361350059405424,-11.868164241239471224,0.040306091310000624617    
if so, where is the 0 position? or how to define the origin point? 

2. How to convert the coordinate from the kitti gps raw data to the example one in KittiGps_converted.txt?   Could you share the process of the conversion? 

Thank you for your help!

varunagrawal, 2020-06-09:
You can set the first location as origin. World frame can be set arbitrarily, but GPS is always latitude, longitude and altimeter.

I am not sure how did you match the example data to that specific subset of the KITTI subset.

mikesheffler, 2020-06-11:
It's some sort of rectangularized coordinate system. Given the magnitude of the coordinates, it's not a big one like UTM, so it's probably a local ENU or NED. Who make the conversion and even what the conversion is seems to have been lost to time.

Check out this [frame from the KITTI 'trailer'](https://youtu.be/KXpZ6B1YB_k?t=54). I haven't processed this data in forever, but I remember that that section on the left that is shaped like Nevada is definitely in there. You could back out the geodetic coordinates from the local rectangular coordinates ... but you already know what the geodetic coordinates are, because you have the truth from the dataset.

You could check the KITTI dataset to see if it's one of [_their_ rectangularizations](http://www.cvlibs.net/datasets/kitti/setup.php), but I'd bet on someone having put it into a format that works for the test they were doing at the moment :/