borglab/gtsam #291: [Feature Request] Add GPU optimizations

acxz, 2020-05-06:
## Feature

It would be great if a GPU accelerated sparse matrix library can be incorporated into GTSAM. These libraries include: AMD ROCM's [rocSPARSE](https://rocmdocs.amd.com/en/latest/ROCm_Libraries/ROCm_Libraries.html#rocsparse) & NVIDIA CUDA's [cuSPARSE](https://developer.nvidia.com/cusparse).

[hipSPARSE](https://rocmdocs.amd.com/en/latest/ROCm_Libraries/ROCm_Libraries.html#hipsparse) is a wrapper around both rocSPARSE and cuSPARSE that allows either one to be used as a backend depending on if you have ROCm or CUDA installed.

## Motivation

Gotta go fast! NVIDIA mentions that its cuSPARSE provides up to 5x speed over CPU alternatives, see: https://developer.nvidia.com/cusparse

## Pitch

Having the option to build GTSAM with hipSPARSE allows GPU accelerated spare matrix calculations to occur with both AMD and NVIDIA devices. This can provide a substantial speedup.

## Alternatives

N/A

## Additional context

It is interesting to note that @dongjing3309's [minisam](https://github.com/dongjing3309/minisam) has the option to use CUDA for its sparse linear solver.

ProfFan, 2020-05-06:
I took a look at minisam, I think they are using cuSPARSE for the Cholesky decomposition step. I think this would be very easy to add to the current codebase, as the only piece of code that needs to be modified is the Cholesky solver (plus some CMake-fu). It should be interesting to benchmark the speedup with some really large datasets.

dellaert, 2020-05-07:
I remember that in fact the CUDA solvers were slower.

varunagrawal, 2020-05-07:
Perhaps Nvidia have improved their cuSparse library significantly by now.

acxz, 2020-06-07:
Came across this: https://developer.nvidia.com/cholmod

Seems like CHOLMOD already has GPU capability built in, we just need a flag to turn it on.

ProfFan, 2020-06-07:
Yes that is on the roadmap :) Should be easy once #111 is merged.

ProfFan, 2020-06-09:
@acxz I have added some other solvers like CHOLMOD and cuSparse in #111. However their performance is not great compared to GTSAM's MULTIFRONTAL_CHOLESKY. There should be a lot of room for optimization though I don't have bandwidth on this currently.

ProfFan, 2020-06-09:
BTW, CHOLMOD's GPU support is somewhat exaggerated: it only supports GPU cholesky with integers, not doubles.

acxz, 2020-06-09:
> added some other solvers like CHOLMOD and cuSparse

Nice! I am sure the optimizations can come later.

> CHOLMOD's GPU support is somewhat exaggerated: it only supports GPU cholesky with integers, not doubles.

rip, makes me sad

ProfFan, 2020-06-09:
> rip, makes me sad

me bursted into tears after wasting one day building custom SuiteSparse with GPU and figuring out how to make it *use* the gpu...

varunagrawal, 2020-07-28:
I spent some time looking into ROCm this past weekend and it looks to be ideal for tackling this issue in a transparent way.

[image]