borglab/gtsam #301: Bug while using dogleg optimizer

kvmanohar22, 2020-05-07:
I have created an unit test (`examples/SFMExample_SmartFactor_bug.cpp`) to illustrate the issue and can be found in my [fork](https://github.com/kvmanohar22/gtsam/tree/dogleg_bug)

In the discussion in [issue:367](https://bitbucket.org/gtborg/gtsam/issues/367/isam2-with-smart-factors-null-ptr-segfault), this bug was resolved but seems like it is popping up again.

## Description

This error seems to pop up when using Dogleg optimizer as part of ISAM2. Following is the reason for this error.

Assume `graph_` represents the existing graph (passed previously to `gtsam::ISAM2`) that has smart factors. Suppose a smart factor was already in the previous graph and we update the smart factor with a new measurement from a pose (this pose was not part of previous graph).

Assuming that the following condition holds true ([line](https://github.com/borglab/gtsam/blob/b4eb0c233d683eaf47511ee9b062dc3c822484df/gtsam/nonlinear/ISAM2.cpp#L424))
```cpp
  // Update delta if we need it to check relinearization later
  if (update.relinarizationNeeded(update_count_))
    updateDelta(updateParams.forceFullSolve);
```
the error can be reproduced. Specifically, the error arises from the following [line](https://github.com/borglab/gtsam/blob/b4eb0c233d683eaf47511ee9b062dc3c822484df/gtsam/nonlinear/ISAM2.cpp#L746) which tries to compute the error of the previous graph but without updating `theta_`

I have tried to simulate this in the unit test.

```cpp
nonlinearFactors_.error(theta_),
```

## Steps to reproduce

Just run the new unit test that I have added (in my fork) using

```bash
make SFMExample_SmartFactor_bug.run
```
which results in the following error

```bash
kv@kv:~/slam/gtsam/build$ make SFMExample_SmartFactor_bug.run
[ 33%] Built target metis
[100%] Built target gtsam
[100%] Building CXX object examples/CMakeFiles/SFMExample_SmartFactor_bug.dir/SFMExample_SmartFactor_bug.cpp.o
[100%] Linking CXX executable SFMExample_SmartFactor_bug
[100%] Built target SFMExample_SmartFactor_bug
Initial Estimates:
Values with 3 values:
Value 0: (N5gtsam5Pose3E) R:
[
 	 0.208654,  0.073369, -0.975233;
    	 0.949229, -0.255245,  0.183888;
    	-0.235432, -0.964089, -0.122902
  ]
[29.8, 0.05, 0.1]';

Value 1: (N5gtsam5Pose3E) R:
[
 	-0.523665,  0.232365, -0.819623;
    	 0.818747, -0.128606, -0.559566;
    	-0.235432, -0.964089, -0.122902
  ]
[21.0364, 21.1071, 0.1]';

Value 2: (N5gtsam5Pose3E) R:
[
 	-0.949229,  0.255245, -0.183888;
    	 0.208654,  0.073369, -0.975233;
    	-0.235432, -0.964089, -0.122902
  ]
[-0.05, 29.8, 0.1]';

Results:
Values with 3 values:
Value 0: (N5gtsam5Pose3E) R:
[
 	-0.0253294,  0.0208786,  -0.999461;
    	  0.998264, -0.0526448, -0.0263988;
    	-0.0531676,  -0.998395, -0.0195089
  ]
[30.0013, -0.000248155, 0.00250433]';

Value 1: (N5gtsam5Pose3E) R:
[
 	  -0.709983,   0.0133575,   -0.704092;
    	   0.704144, -0.00108604,   -0.710056;
    	 -0.0102493,    -0.99991, -0.00863457
  ]
[21.2138, 21.2135, 0.00132106]';

Value 2: (N5gtsam5Pose3E) R:
[
 	 -0.999205, -0.0254096,  0.0307042;
    	-0.0320221,  0.0531939,  -0.998071;
    	 0.0237273,  -0.998261, -0.0539653
  ]
[-2.03514, 30.1004, 1.98536]';

terminate called after throwing an instance of 'gtsam::ValuesKeyDoesNotExist'
  what():  Attempting to at the key "3", which does not exist in the Values.
Aborted (core dumped)
examples/CMakeFiles/SFMExample_SmartFactor_bug.run.dir/build.make:57: recipe for target 'examples/CMakeFiles/SFMExample_SmartFactor_bug.run' failed
make[3]: *** [examples/CMakeFiles/SFMExample_SmartFactor_bug.run] Error 134
CMakeFiles/Makefile2:2660: recipe for target 'examples/CMakeFiles/SFMExample_SmartFactor_bug.run.dir/all' failed
make[2]: *** [examples/CMakeFiles/SFMExample_SmartFactor_bug.run.dir/all] Error 2
CMakeFiles/Makefile2:2667: recipe for target 'examples/CMakeFiles/SFMExample_SmartFactor_bug.run.dir/rule' failed
make[1]: *** [examples/CMakeFiles/SFMExample_SmartFactor_bug.run.dir/rule] Error 2
Makefile:884: recipe for target 'SFMExample_SmartFactor_bug.run' failed
make: *** [SFMExample_SmartFactor_bug.run] Error 2

```

## Expected behavior
The error should not have popped up

## Environment
OS: Ubuntu 16.04
Language: C++, custom build

## Additional information
This bug was discussed previously in the following discussion. [link](https://bitbucket.org/gtborg/gtsam/issues/367/isam2-with-smart-factors-null-ptr-segfault), specifically comment by Sean Bowman.

kvmanohar22, 2020-05-08:
Ping @dellaert

dellaert, 2020-05-08:
I have zero time to look at this now, but if you have the time to debug and propose a fix, I would gladly review.

dellaert, 2020-05-08:
By the way, search through previous issues, because we had a similar issue a few months back I think

dellaert, 2020-05-08:
As I see you already discovered :-)

varunagrawal, 2024-01-08:
So I managed to figure this one out. The issue is that `nonlinearFactors_.error(theta_)` is computed as an argument to `DoglegOptimizerImpl::Iterate`, where a factor with key `3` exists but `theta_` has not been updated with the new values yet. 

This is because in `ISAM2::update`, the delta update is computed first to check for relinearization before the new factors and Values are added. The delta update calls the above error method with missing keys in `theta_` leading to the issue.

varunagrawal, 2024-01-08:
@kvmanohar22 to your point, you identified the issue correctly. All we need to do is call `addVariable(newTheta)` before computing the delta.

PR #1707 (2024-08-25): ISAM2: Add variables before computing delta
Fixes #301 

A `gtsam::ValuesKeyDoesNotExist` error is thrown from the function `nonlinearFactors_.error(theta_)`.
The issue is that `nonlinearFactors_.error(theta_)` is computed as an argument to `DoglegOptimizerImpl::Iterate`, where a factor with the new key exists but `theta_` has not been updated with the new key-value pairs yet.

It is a simple quirk that Dogleg requires the updated Values since it needs them to compute the delta update via the dogleg point.

The issue in `ISAM2::update` is that the delta update is computed first to check for re [... cut]
closed 2024-08-25 by varunagrawal