borglab/gtsam #306: Failed test on ARM 32bit architecture

jlblancoc, 2020-05-10:
## Description

One unit test fails to pass in `armhf`

```
186/234 Test #186: testQPSolver ...........................***Failed    0.04 sec
Not equal:
expected:
4.371875
actual:
290687824935850081763694373348985159914898047944127280960733468936086603046127808004981004843733268589327788990324383899071870077623672891177494692327651468166411650774643726713810200573276388989574637258303295240952023116603342313435955200.000000
/<<PKGBUILDDIR>>/gtsam_unstable/linear/tests/testQPSolver.cpp:330: Failure: "assert_equal(0.437187500e01,error_actual, 1e-7)" 
There were 1 failures 
```

## Steps to reproduce

1. Build and run tests on architectures != amd64.

## Expected behavior

All unit tests should pass! :-)

## Environment

See build logs in:
https://launchpad.net/~joseluisblancoc/+archive/ubuntu/gtsam-develop/+packages

Note that right now, tests can't run for `ppc64el` and `arm64`, but errors there seem to be related to limited RAM in the builders instead.

jlblancoc, 2020-05-14:
Progress: I traced this down to a problem parsing QPS files. 

For instance, in 64bit builds the file [examples/Data/QPTEST.QPS](https://github.com/borglab/gtsam/blob/develop/examples/Data/QPTEST.QPS) is correctly parsed as:
``` 
QPTEST.QPS: 
Quadratic cost factors: 
size: 1
factor 0: 
 keys: X1(1) X2(1) 
Augmented information matrix: [
	   8,    2, -1.5;
 	   2,   10,    2;
 	-1.5,    2,    0
]
...
```

while in i386 it's read as:
```
QPTEST.QPS 
Quadratic cost factors: 
size: 1
factor 0: 
keys: X1(1) X2(1) 
Augmented information matrix: [
     8,            2,         -1.5;
     2,           10,            2;
  -1.5,            2, **2.36234e+111**
]
...
```

I'm not familiar with the QPS file format... any pointers? :-)

jlblancoc, 2020-05-14:
PS: it totally looks like an uninitialized vector somewhere, pretty much like those recently fixed by @ProfFan ...

dellaert, 2020-05-14:
Valgrind say anything?

dellaert, 2020-05-14:
@jlblancoc many many thanks for tracking this down !

ProfFan, 2020-05-14:
Valgrind probably won't say anything here as this is the de factor behavior of Eigen... However I would guess it's where we initialize the Augmented information matrix. For example, a default constructor call/a "static" ( `Matrix3 xxx;` like) declaration of a matrix.

jlblancoc, 2020-05-15:
This is fixed now, Yay! 
It was this tiny thing: b73f03646d4c682399ad6304b03ffa6645858088

Fixed by PR #310

jlblancoc, 2020-06-19:
Reopening since the former proposed solution was actually a workaround, a better solution would be to fix the construction of the quadratic cost factor in the test referenced above.

varunagrawal, 2020-11-19:
I'm seeing this issue on GCC compilers (Clang works fine) and a 64-bit architecture. I'd like to fix this before the weekend.

ProfFan, 2020-11-19:
It's 100% UB... UBSan didn't report anything so I think you need to manually scan for it :)

varunagrawal, 2020-11-19:
Here's some more info:

I tried commenting out the other tests so that I could isolate the issue better. Funnily enough, the test in question passes, when you comment out the other tests. I tracked down the culprit and it seems like the `CHECK` in `TEST(QPSolver, HS268)` is causing some weirdness, because if I only comment out that one line, everything works fine. Even on inspecting the Hessian factors, it is correctly initialized.

Any thoughts?

varunagrawal, 2020-11-19:
Found the issue. It's the damn parser not reading the constant term correctly. _sigh_

varunagrawal, 2020-11-19:
The fix is the dumbest thing: In `QPSVisitor` force `f = 0` by explicitly initializing it.

varunagrawal, 2020-11-19:
@jlblancoc can you please test the fix in #607 on the arm builds?

PS a few PRs more and we could have had the fix in 306*2 = 612. :joy:

varunagrawal, 2020-12-01:
@jlblancoc gentle reminder.

jlblancoc, 2020-12-01:
copy that...

jlblancoc, 2020-12-02:
Yay, [it's fixed](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/gtsam-develop/+packages)! :+1:

jlblancoc, 2020-12-02:
Confirmed: #306 closed via #607

closed 2020-12-02 by jlblancoc

varunagrawal, 2020-12-02:
Yes!!! Awesome

varunagrawal, 2020-12-02:
We can now experiment with embedded ARM platforms. 😁

commit b73f0364 (2020-05-15): Fix potential uninitialized matrix elements (Fixes #306)
PR #310 (2020-05-15): Fix 32-bit unit test failures
Closes #308 
Closes #306 

Don't merge until [these PPA builds](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/sandbox/+packages) end, to confirm that the test errors have been fixed.

Update: It can be merged now, PPA builds pass!

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/310)
PR #607 (2020-11-19): Enforce constant term in QP to be default 0
Fixes #306.

I also added braces for the `if` statements because that is good style.