borglab/gtsam #308: Failed tests in 32bit builds (i386 on PPA build farm)

jlblancoc, 2020-05-11:
## Description

Failure 1: (this is a weird one, but must have an explanation ?)

```
130/234 Test #130: testAHRSFactor .........................***Failed    0.01 sec
not equal:
expected = [
	0, 0, 0;
 	0, 0, 0;
 	0, 0, 0
]
actual = [
	        -1.01,  -4.92831e-85, -5.82588e-280;
 	 -4.92831e-85, -2.30763e-169, -7.25981e-280;
 	-9.12492e-102,  3.40735e-171, -2.54655e-280
]
actual - expected = [
	        -1.01,  -4.92831e-85, -5.82588e-280;
 	 -4.92831e-85, -2.30763e-169, -7.25981e-280;
 	-9.12492e-102,  3.40735e-171, -2.54655e-280
]
/<<PKGBUILDDIR>>/gtsam/navigation/tests/testAHRSFactor.cpp:376: Failure: "assert_equal(expectedDelRdelBiasOmega, preintegrated.delRdelBiasOmega(), 1e-3)"
```

Failure 2: (this one can be fixed by just relaxing a bit the thresholds)

```
 30/234 Test  #30: testLine3 ..............................***Failed    0.01 sec
Not equal:
expected:

  A[1] = [
	       0        0  5.00001 -5.00001;
   	-7.07107 -7.07107        0        0
  ]
  b = [  0 -0 ]
  No noise model
actual:

  A[1] = [
	       0        0        5       -5;
   	-7.07107 -7.07107        0        0
  ]
  b = [  0 -0 ]
  No noise model
not equal:
expected = [
	       0,        0,  5.00001, -5.00001;
 	-7.07107, -7.07107,        0,        0
]
actual = [
	       0,        0,        5,       -5;
 	-7.07107, -7.07107,        0,        0
]
actual - expected = [
	           0,            0, -8.58159e-06,  8.58159e-06;
 	-6.97755e-07, -6.97755e-07,            0,            0
]
Mismatch in Jacobian 1 (base 1), as shown above
/<<PKGBUILDDIR>>/gtsam/geometry/tests/testLine3.cpp:149: Failure: "gtsam::internal::testFactorJacobians(name_, f, val, 1e-5, 1e-7)" 
There were 1 failures
```

## Steps to reproduce

1. Run the tests on 32bit Intel architecture. 

## Expected behavior

Tests should pass. 

## Environment

https://launchpad.net/~joseluisblancoc/+archive/ubuntu/gtsam-develop/+packages

ProfFan, 2020-05-11:
I fixed the first one (but only in the feature/new_wrapper branch for the new wrapper...Maybe you can cherry-pick that into `develop`?

https://github.com/borglab/gtsam/pull/259/commits/b8fa9d5cec35e5d207cffd874999c2881cedf398

jlblancoc, 2020-05-11:
@dellaert would you like a cherry pick of that commit + a fix for the other test in an independent pr?

dellaert, 2020-05-11:
Feel fee to combine - whatever you think best ;-)

PR #310 (2020-05-15): Fix 32-bit unit test failures
Closes #308 
Closes #306 

Don't merge until [these PPA builds](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/sandbox/+packages) end, to confirm that the test errors have been fixed.

Update: It can be merged now, PPA builds pass!

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/310)
closed 2020-05-15 by dellaert