borglab/gtsam #311: Demangle Type Info

varunagrawal, 2020-05-17:
## Feature

Currently GTSAM outputs type information like this

```
N5gtsam12GenericValueIN5Eigen6MatrixIdLi4ELi1ELi0ELi4ELi1EEEEE
```

which makes debugging hard.

It would be great to have GTSAM print type info like this instead

```
gtsam::GenericValue<Eigen::Matrix<double, 4, 1, 0, 4, 1> >
```

which is actually quite easily doable.

## Motivation

Make understanding and debugging much, much easier.

## Pitch

This feature is very straightforward. The code to enact this change can be found [here](https://stackoverflow.com/a/4541470/1236990). I have tried it out on a local version of GTSAM and it works great.

I was originally getting this error message:

```
Failure: "Exception: Attempting to retrieve value with key "X0", 
type stored in Values is N5gtsam12GenericValueIN5Eigen6MatrixIdLi4ELi1ELi0ELi4ELi1EEEEE 
but requested type was N5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEE" 
```

and now I get this

```
Failure: "Exception: Attempting to retrieve value with key "X0", 
type stored in Values is gtsam::GenericValue<Eigen::Matrix<double, 4, 1, 0, 4, 1> > 
but requested type was Eigen::Matrix<double, -1, -1, 0, -1, -1>" 
```

which makes it clear immediately what the issue is.

## Alternatives

N/A

## Additional context

ProfFan, 2020-05-17:
I always use `c++filt`: `make check | c++filt` :)

varunagrawal, 2020-05-17:
But why would you need the user to use that filter when we can support it directly? :)

varunagrawal, 2020-05-19:
I also expect the fix to this to help in print statements in the wrappers. Now that would be cool!

closed 2020-05-20 by varunagrawal

TheSeanParker, 2024-06-21:
> ## Feature
>
> Currently GTSAM outputs type information like this
>
> ```
> N5gtsam12GenericValueIN5Eigen6MatrixIdLi4ELi1ELi0ELi4ELi1EEEEE
> ```
>
> which makes debugging hard.
>
> It would be great to have GTSAM [...]

so great work!
thanks!

PR #313 (2020-05-20): Demangle the type when printing
This PR fixes #311.

I expect to get quite a bit of feedback regarding this.

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/313)