borglab/gtsam #319: numericalDerivative returning unexpected jacobians for the Pose2 L2-norm error

acxz, 2020-05-23:
## Description

When computing the numerical Jacobians (w.r.t x1, x2, y1, y2, theta1, theta2) of the following equation: sqrt((x2 - x1)^2 + (y2 - y1)^2) I obtain a different result from the analytical Jacobians.
The error seems to only show up when I have a nonzero theta1 or theta2 value.

With:
```
    double x1 = 0;
    double y1 = 0;
    double theta1 = 0;

    double x2 = 0;
    double y2 = 1.2;
    double theta2 = M_PI/6;
```
The jacobians result in: (expected is from `numericalDerivative.h` and actual is the analytical expression I have computed)
```
not equal:
expected = [
 	     0.5 0.866025        0
  ]
actual = [
 	0 1 0
  ]
actual - expected = [
 	    -0.5 0.133975        0
  ]
```

Here is the code I used to test this bug: https://gist.github.com/acxz/283a65b414061f6f962c654d33087ffc

## Steps to reproduce

1. Compile and run linked test-numdiff.cpp file (CMakeLists.txt provided)

## Expected behavior

The numerical derivative for the L2-norm: sqrt((x2 - x1)^2 + (y2 - y1)^2) should not be dependent on the theta values.

I tried to follow through the central differencing code, but I got lost on this line: 
https://github.com/borglab/gtsam/blob/develop/gtsam/base/numericalDerivative.h#L147

As in this line gave me a different `dy1` than I had expected when doing the central differencing by hand.
After getting stuck there I gave up debugging and decided to post the issue here to see if anyone could reproduce/take a look at this. 

## Environment

OS: Arch Linux
Language: C++
gcc: 10.1.0
develop branch of gtsam

## Additional information

I am fairly certain I have computed the analytical Jacobians correctly, but if there is a user mistake on my end, I apologize.

ProfFan, 2020-05-23:
I believe the reason is that the derivative here is actually lying on the SE(2) manifold, and the operation you are doing here (with the numericalDerivative) is cutting off a `2x1` slice from the `3x1` actual Jacobian on SE(2). The derivative you computed manually is on `R^2 x SO(2)`, and they have different tangent representations.

ProfFan, 2020-05-23:
Again, please correct me if not :)

ProfFan, 2020-05-23:
For a code representation, I think this snippet from `Pose2.cpp` actually considered this discrepancy:
```c++
/* ************************************************************************* */
double Pose2::range(const Point2& point,
    OptionalJacobian<1,3> Hpose, OptionalJacobian<1,2> Hpoint) const {
  Point2 d = point - t_;
  if (!Hpose && !Hpoint) return d.norm();
  Matrix12 D_r_d;
  double r = norm2(d, D_r_d);
  if (Hpose) {
      Matrix23 D_d_pose;
      D_d_pose << -r_.c(),  r_.s(),  0.0,
                  -r_.s(), -r_.c(),  0.0;
      *Hpose = D_r_d * D_d_pose;
  }
  if (Hpoint) *Hpoint = D_r_d;
  return r;
}
```

dellaert, 2020-05-23:
Indeed, please read math.pdf to get some background on this discrepancy. Believe the numerical derivatives, as they take into account the correct retractions on the manifold.

If after this, you are still convinced there is an issue, let’s talk more, but know that it can be counterintuitive.

I propose to close this issue for now, but feel free to reopen in the latter case.

closed 2020-05-23 by dellaert

acxz, 2020-05-24:
Thanks for clearing that up and pointing me in the right direction. Looks like I need to do some reading. :)