borglab/gtsam #324: Python Examples

soulslicer, 2020-05-27:
The `examples` folder is useful but the code is in C++. Is there a Python Equivalent for all the examples in that folder? I see that there are for `matlab` but I couldnt find the location of the Python ones

varunagrawal, 2020-05-27:
There aren't any plans to port all the examples currently since 

1. There is a 1-to-1 equivalence between the C++ and the Python examples (for the most part).
2. We are working on a new Python wrapper which will be much cleaner, hence this would be just duplicated effort.

It is possible to create a Python version of the StereoVO examples without too much effort. It could also be viewed as a good pedagogical exercise IMO.

soulslicer, 2020-05-27:
Sure. What kind of timeline is there for the new Python wrapper. Will it use pybind11

varunagrawal, 2020-05-27:
Yes it will be using Pybind11 (which is awesome).

@ProfFan can give you a better idea of the timeline.

ProfFan, 2020-05-28:
@soulslicer If you want to use it for non-production environments (a.k.a. research), just clone the `feature/new_wrapper` branch and compile with `-DGTSAM_TYPEDEF_POINTS_TO_VECTORS=ON -DGTSAM_BUILD_PYTHON=ON`. after that, go to the build folder and type `python3 setup.py install`.

zwn, 2020-05-28:
The commit 26d6cb3d6ebc06b95270649ab3e084a80adade66 broke some of the examples that use `plot_pose3` function - the third argument is now covariance matrix while before it was axis length:

<pre>$ grep -R plot_pose3 gtsam/examples/
<font color="#75507B">gtsam/examples/ImuFactorExample.py</font><font color="#06989A">:</font>from gtsam.utils.plot import <font color="#EF2929"><b>plot_pose3</b></font>
<font color="#75507B">gtsam/examples/ImuFactorExample.py</font><font color="#06989A">:</font>            <font color="#EF2929"><b>plot_pose3</b></font>(POSES_FIG, pose_i, 0.1)
<font color="#75507B">gtsam/examples/Pose3SLAMExample_g2o.py</font><font color="#06989A">:</font>        plot.<font color="#EF2929"><b>plot_pose3</b></font>(1, resultPoses.atPose3(i))
<font color="#75507B">gtsam/examples/PreintegrationExample.py</font><font color="#06989A">:</font>from gtsam.utils.plot import <font color="#EF2929"><b>plot_pose3</b></font>
<font color="#75507B">gtsam/examples/PreintegrationExample.py</font><font color="#06989A">:</font>        <font color="#EF2929"><b>plot_pose3</b></font>(POSES_FIG, actualPose, 0.3)
<font color="#75507B">gtsam/examples/PreintegrationExample.py</font><font color="#06989A">:</font>                <font color="#EF2929"><b>plot_pose3</b></font>(POSES_FIG, predictedNavState.pose(), 0.1)
<font color="#75507B">gtsam/examples/ImuFactorExample2.py</font><font color="#06989A">:</font>        gtsam_plot.<font color="#EF2929"><b>plot_pose3</b></font>(fignum, pose_i, 10)
<font color="#75507B">gtsam/examples/VisualISAM2Example.py</font><font color="#06989A">:</font>        gtsam_plot.<font color="#EF2929"><b>plot_pose3</b></font>(fignum, pose_i, P=None, axis_length=10)
</pre>

As you can see I have locally adjusted `gtsam/examples/VisualISAM2Example.py` so it works. 

Should fixing that wait for the new python wrapper as well or is that an orthogonal issue?

ProfFan, 2020-05-28:
You can try make a PR I think :)

dellaert, 2020-05-28:
> You can try make a PR I think :)

Actually I think @varunagrawal should respond here and possibly fix, if needed. We can't break the API - and should have caught it in that PR.

varunagrawal, 2020-05-28:
It's *technically* not broken because `axis_length` is a keyword argument, but it's my fault for not updating all the examples. Will update by COB today.

zwn, 2020-05-28:
Actually `axis_length` is both keyword and positional argument and the examples happen to use the positional variant. Did you mean to say _keyword only argument_? But for that the signature would have to look different:

```python
def plot_pose3(fignum, pose, P, *, axis_length=0.1):
```

varunagrawal, 2020-05-28:
Isn't that the signature already? From what I've seen in the commit, `axis_length` always has a default value and follows positional arguments. I just need to make it a keyword arg in the examples and everything should work.

zwn, 2020-05-28:
Please note the addition of the asterix to the signature. For more explanation please see https://www.python.org/dev/peps/pep-3102/ as a way to force the caller to always specify the argument with a keyword.

zwn, 2020-05-29:
> You can try make a PR I think :)

@ProfFan: Please see #326.

dellaert, 2020-05-30:
Btw, there is a readme with status of porting python examples. Feel free to port more of them and contribute PRs :-)

closed 2020-05-30 by dellaert