borglab/gtsam #333: Unit tests: cover cmake-based building of user program

jlblancoc, 2020-06-02:
## Feature

It would be great to add a new Unit Test consisting of invoking cmake to build a simple program using the just-generated `gtsam*-config.cmake` + `gtsam*-targets.cmake`, etc.

## Motivation

Detect bugs regarding exported targets for different build configurations, usage of embedded vs system libraries etc. 

## Additional context

Related to: #292, #332

tmcg0, 2020-06-04:
Could this be covered in CI? Thinking about the number of environments it would be necessary to test against.

varunagrawal, 2020-06-04:
This would be awesome and a separate "test suite" I guess? We already have a Matlab wrapper Hello World example, adding a python and C++ version and just having CMake build and call those (very doable from experience) would be of tremendous value.

jlblancoc, 2020-06-04:
> Could this be covered in CI?

At least testing it in one or two systems/configurations would be great :-)
As @varunagrawal says, "all" that is required is to call cmake a couple of times and check its process numeric result.