borglab/gtsam #342: Serialization to XML File missing a line

gchenfc, 2020-06-09:
## Description
The function `serializeToXMLFile` in `gtsam/base/serialization.h` is missing a the last line of the XML file which causes it to not parse correctly on deserialize.  The missing line is 
`</boost_serialization>`.

## Steps to reproduce

1. Create a factor graph.
2. Create an XML file using `serializeToXMLFile`.  Notice that the `<boost_serialization ...>` tag is not closed.
3. Create an XML file using `serializeXML` and write the output to a file.  Notice that the `<boost_serialization ...>` tag **is** closed this time.

Unit test (sorry it's not stripped all the way down to MWE, but it's pretty barebones in any case):
```cpp
/**
 * @file testSerialization.cpp
 *
 * @brief Simple test for Pose2-type factors used in planar-collab-nav-sim
 *
 * @author Gerry Chen
 */

#include <gtsam/base/serialization.h>
#include <gtsam/geometry/Pose2.h>
#include <gtsam/linear/NoiseModel.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/sam/RangeFactor.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/base/serializationTestHelpers.h>

#include <stdio.h>

#include <CppUnitLite/TestHarness.h>

using namespace std;

// GUID declarations are needed for serialize to work.  Add more as needed.
BOOST_CLASS_EXPORT_GUID(gtsam::PriorFactor<gtsam::Pose2>,
                        "gtsam::PriorFactorPose2");
BOOST_CLASS_EXPORT_GUID(gtsam::BetweenFactor<gtsam::Pose2>,
                        "gtsam::BetweenFactorPose2");
BOOST_CLASS_EXPORT_GUID(gtsam::RangeFactor<gtsam::Pose2>,
                        "gtsam::RangeFactorPose2");
// Note: noise model constructors may automatically specialize to more derived
// type
BOOST_CLASS_EXPORT_GUID(gtsam::noiseModel::Gaussian,
                        "gtsam_noiseModel_Gaussian");
BOOST_CLASS_EXPORT_GUID(gtsam::noiseModel::Diagonal,
                        "gtsam_noiseModel_Diagonal");
BOOST_CLASS_EXPORT_GUID(gtsam::noiseModel::Isotropic,
                        "gtsam_noiseModel_Isotropic");
BOOST_CLASS_EXPORT_GUID(gtsam::noiseModel::Unit, "gtsam_noiseModel_Unit");

GTSAM_VALUE_EXPORT(gtsam::Pose2);

/* ************************************************************************* */
TEST( testSerialization, SimplePose2Example ) {
  gtsam::Pose2 btwn;
  auto range_noise =
      gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, 1, 1>(0.1));
  auto odom_noise = gtsam::noiseModel::Diagonal::Sigmas(gtsam::Vector3::Ones());
  auto prior_noise = gtsam::noiseModel::Diagonal::Sigmas(Eigen::Vector3d(0.2,
  0.1, 0.1));

  gtsam::NonlinearFactorGraph graph;
  graph.emplace_shared<gtsam::PriorFactor<gtsam::Pose2>>(gtsam::Key(0), btwn,
  prior_noise);
  graph.emplace_shared<gtsam::BetweenFactor<gtsam::Pose2>>(
      gtsam::Key(0), gtsam::Key(1), btwn, odom_noise);
  graph.emplace_shared<gtsam::RangeFactor<gtsam::Pose2>>(
      gtsam::Key(0), gtsam::Key(2), 2, range_noise);

  // using serializeToXMLFile
  gtsam::serializeToXMLFile(graph, "testSerialization.txt");
  // read back
  std::ifstream ifs("testSerialization.txt");
  std::string actual( (std::istreambuf_iterator<char>(ifs) ),
                       (std::istreambuf_iterator<char>()    ) );

  // using serializeXML and writing string to file
  gtsam::NonlinearFactorGraph actual;
  std::string expected = serializeXML(graph);
  // write out to file so that we can use diff tools
  ofstream ofile;
  ofile.open ("testSerialization2.txt");
  ofile << expected;
  ofile.close();

  // print out serializations to console
  cout << "<<<<<<<<<<<<<<<<<<" << endl;
  cout << expected << endl;
  cout << "==================" << endl;
  cout << actual << endl;
  cout << ">>>>>>>>>>>>>>>>>>" << endl;
  cout << thingy << endl;

  // these are not equal
  EXPECT(expected == actual);
}

/* ************************************************************************* */
int main() { TestResult tr; return TestRegistry::runAllTests(tr); }
/* ************************************************************************* */
```

This creates 2 files: `testSerialization.txt` and `testSerialization2.txt`.  You can use a diff tool to see the difference.

## Expected behavior
`serializeToXMLFile` should write this line at the end of the file:
`</boost_serialization>`

## Environment

MacOS, C++, gtsam develop branch from about a week ago (commit 106cab5b53181d5c8cccecbbddca1e9ffaf35a20), Apple clang 11.0.3

## Additional information
I assume a buffer isn't getting flushed or something, but I can't really follow what the `boost::serialization::make_nvp` is doing and if the issue is in the way gtsam is handling the file streams or if it's within boost itself.

ProfFan, 2020-06-09:
Strange, there should be round-trip tests for XML serializer in the testset already...

ProfFan, 2020-06-09:
NVM, the file variant is not fixed but the string one is:
```
template<class T>
std::string serializeXML(const T& input, const std::string& name="data") {
  std::ostringstream out_archive_stream;
  // braces to flush out_archive as it goes out of scope before taking str()
  // fixes crash with boost 1.66-1.68
  // see https://github.com/boostorg/serialization/issues/82
  {
    boost::archive::xml_oarchive out_archive(out_archive_stream);
    out_archive << boost::serialization::make_nvp(name.c_str(), input);
  } /// HERE!!!!! Need to duplicate the the serializeToFile variant !!!!!
  return out_archive_stream.str();
}
```

@gchenfc Could you make a PR fixing this?

closed 2020-06-25 by varunagrawal