borglab/gtsam #344: Nightly build of Python package

varunagrawal, 2020-06-09:
## Feature

GTSAM is currently available through `pip` over PyPI, but given the steady stream of improvements, the current version is lacking a lot of features. It would be nice to have a nightly build which publishes the daily snapshot of the repo as a package which can be installed easily.

## Motivation

Having a nightly build would make access to new features and bugfixes a lot more accessible, without the need to wait for a new release.

## Pitch

## Alternatives

Manually compile from source each and every time.

## Additional context

ProfFan, 2020-06-09:
#247 is my attempt to solve this but is blocked :(

varunagrawal, 2020-06-09:
Where is it blocked? I can help if you're preoccupied.

ProfFan, 2020-06-09:
Adding a bot user and give it access to GTSAM :)

varunagrawal, 2020-06-09:
Won't we need Frank to give access to the bot? I don't believe I have access.

ProfFan, 2020-06-09:
Yeah thus it stalled... Otherwise we need to manually trigger the build...

dellaert, 2020-06-09:
I was told the bot was for deleting branches, which I do not agree with. If you want me to give the bots access, let me know the "deleting branches bot" is not running.

ProfFan, 2020-06-09:
@dellaert No that is not happening :)

Long story short:

1. The bot only need admin access to `gtsam-manylinux-build` not `gtsam`
2. the personal access token (representing the bot) is then written into the secret entry `PYTHON_CI_REPO_ACCESS_TOKEN` (in settings of GTSAM repo, by @dellaert )
3. after #247  is merged, the workflow in `gtsam` repo will call the `gtsam-manylinux-build` repo to start the github actions, building the nightly wheels

ProfFan, 2020-06-09:
Also, the token will not be leaked to untrusted third parties because GitHub Secrets will only allow PRs originating from the `gtsam` repo to access the secret, not forks.

dellaert, 2020-06-09:
I would like @varunagrawal  to comment, because the only thing that I got was a fairly general authorization request for any type of bot thing. If he does not respond can you trigger a new authorization request for the more restricted activity you just outlined above?

varunagrawal, 2020-06-09:
> I was told the bot was for deleting branches, which I do not agree with. If you want me to give the bots access, let me know the "deleting branches bot" is not running.

This was for another bot, the idea of which I canned after seeing your email. My mistake, sorry for the confusion.

closed 2020-06-12 by ProfFan

ProfFan, 2020-06-12:
@varunagrawal Note that nightlies now live in https://github.com/borglab/gtsam-manylinux-build/actions :)