borglab/gtsam #357: release bump

dellaert, 2020-06-18:
I created a new release branch to cement things before we switch to 4.1
4.1. will remove the cython wrapper.

@ProfFan  will bump release number
Which PR's *need* to be merged in 4.0.3? @jlblancoc @mikesheffler @varunagrawal @acxz

varunagrawal, 2020-06-18:
I want to definitely get #343 and #358 in. #358 can happen since this is a version bump. :tada:

dellaert, 2020-06-18:
> I want to definitely get #343 and #358 in. #358 can happen since this is a version bump. 🎉

No, sorry, this is a *minor* version bump *before* we version bump to 4.1, which can include API changes. 4.0.3. should not have API changes. Does that change anything?

varunagrawal, 2020-06-18:
Ah pardon me. #358 should be for 4.1 then.

jlblancoc, 2020-06-19:
> Which PR's need to be merged in 4.0.3? @jlblancoc 

You can merge #314 after my last commit (after CI does its job), which actually finishes the undo of #310 .... as we discussed, #310 changed the behavior in a very core class just to fix an error somewhere in `gtsam_unstable` (recall, the error was only revealed when building in intel/arm 32bit architectures), which we should address someday but is not urgent IMO.

acxz, 2020-06-20:
None of my PRs are essential at the moment.

For using upstream METIS, we need to wait for METIS to expose more of their API with the default installation. (not critical at all)

For TBB we are using deprecated functionality and deprecation warnings do clutter the build, but most users shouldn't see them and it will only be a problem when that functionality is actually removed in later TBB releases.
While I would like to cleanup my TBB PR, many unit tests fail right now and I don't have to time (in the foreseeable future) to fix it up for 4.0.3

varunagrawal, 2020-07-07:
I discussed this minor release with @ProfFan today and this is the plan we've come up with:

1. Identify issues/PRs that we would like to see addressed in this release.
2. Set a date to finalize the release.
3. Freeze all the issues/PRs we wish to push in this release.
4. Address the task items and prepare the release.

I have already made a list in the main comment of all the issues and PRs that would be *nice to have* addressed. We can limit the scope as per priority and requirements.

dellaert, 2020-07-07:
We left “nice to have” behind last week :-) Let’s do the absolute essential, like the broken matlab wrapper. Then pybind wrapper upgrades to 4.1. Everything else can go into 4.1.1.

Speed !!!

varunagrawal, 2020-07-07:
Gotcha! I've updated the list to include the issues that are easy to fix or are serious bugs. I believe we can knock those out by the end of the week.

dellaert, 2020-07-07:
> Gotcha! I've updated the list to include the issues that are easy to fix or are serious bugs. I believe we can knock those out by the end of the week.

Forget “easy to fix”. Pare down even more to *BROKEN NEEDS TO BE FIXED*.

ProfFan, 2020-07-11:
Also , we need to remove `wrap/pybind11` in the release branch.

ProfFan, 2020-07-22:
Yay!

closed 2020-07-22 by ProfFan