borglab/gtsam #369: IMU information in ImuFactorExample

aspears1935, 2020-06-24:
## Feature

If possible, can you please note the type of IMU you used for this example so it will be easier for users to translate the example noise sigma values to what they should use for their own IMUs? I see some old discussions of how to obtain these values for different IMUs, but don’t see any conclusive equations.

## Motivation

Knowing the IMU you used can help us back out how those example numbers were obtained so we can figure out how to obtain the correct values for other IMUs. Thank you.

varunagrawal, 2020-06-24:
I am actually working on this right now and any help would be appreciated.
Can you please point me to the discussions?

mikesheffler, 2020-06-24:
@varunagrawal Here is [one time I waded in on the Google group mailing list](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/gtsam-users/o9brW87ZjZ0/dWOigRcjBAAJ) on the topic.

One thing that is maybe not ideal about the status quo is that a lot of people jump onto the Kitti example as the first thing they try to understand (I did), and the IMU data has no documentation. My memory is that the IMU used in the Kitti tests was a [OXTS RT3003](https://www.oxts.com/app/uploads/2018/02/rtman.pdf). It's also my memory that I couldn't definitively connect the parameters used in the example to the datasheet. I think I got to the point where I realized that the noise numbers might have been an integer multiple of the datasheet value if you squinted just right and interpreted the datasheet in a way that I didn't think was correct. The bias noise were then integer multiples of the original noise numbers, which was not a crazy choice, given that the Allan variance plots aren't available for that IMU (I don't think).

Who converted the data and how is probably lost to the mists of time. 🙁

aspears1935, 2020-06-24:
https://github.com/borglab/gtsam/issues/213
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model

In some of the papers about the new IMU factor, the ADIS16448 IMU is mentioned, but the datasheet doesn't seem to correspond.

mikesheffler, 2020-06-24:
I don't know if this helps, but there is a [nice thread on a (closed) issue on the Kalbr site](https://github.com/ethz-asl/kalibr/issues/63)  about some of the gymnastics that get you from one conception of the data to another. It also notes that they use different parameters for the 16448 than they used to.

aspears1935, 2020-06-25:
Ok. Thanks. I will try to play around with the equations and see if I can match the numbers to any of the 16448 values to see if that might be what was used.

andre-nguyen, 2020-06-30:
I'd just like to chime in here and mention a few things which may or may not help:

1. It's probably best to just go ahead and plot out the allan deviation curves instead of just relying on the datasheet as a sanity check.
2. Some datasheets will give you the required values and it is relatively easy to convert the units to the expected ones in the Kalibr noise model some don't.
3. Unless you have a good calibration setup (temperature controlled chamber, vibration isolated), you're basically going to be guesstimating the values anyways. This is especially true for the random walk part of the model where you're supposed to fit a +1/2 slope to (sort of) the right side of the plot. To do so you need to capture values for higher values of tau (10^3, 10^4, 10^5). Now assuming you capture data for 12 hours that would only give you 4-5 data points for tau = 10^4 resulting in very noisy graphs. But for a 12 hour timespan of data collection (as well as larger spans) you're bound to be affected by your building's thermostat or air currents. Note that this may not be true for decent industrial IMUs which ship with temperature compensation. 
4. I've seen a lot of examples floating around where they assume that the noise values are the same for each axis but this may not be true either, consider that internally some IMUs are composed of multiple mems chips its possible for each chip to have different noise properties. A great example is the xsens boards where you can literally see the two extra perpendicular boards holding the extra sensors.

[image]

varunagrawal, 2020-08-11:
So the numbers from the OXTS don't match up. I guess we have to ask @lucacarlone directly...

lucacarlone, 2020-08-14:
@varunagrawal : I would check this thread about the measurement units required to instantiate the quantities in the IMU factor: https://github.com/borglab/gtsam/issues/213#issuecomment-656958996. 
The IMUFactorExample went through many iterations and it is not easy for me to backtrack the numbers we are using there. 
You can also find an example where we use the IMU stats from the Euroc dataset (https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) in Kimera (https://github.com/MIT-SPARK/Kimera)