borglab/gtsam #380: libmetis-gtsam issue

CrazyPopLin, 2020-07-08:
Hi, I built gtsam follow the [quick start](https://github.com/borglab/gtsam#quickstart) and build the github branch successfully with a install path.

Then I try to build the **OdometryExample.cpp** with cmake(with ROS but plain gtsam). The CMakeLists.txt is shown below. The build was successfully, too. 
```s
cmake_minimum_required(VERSION 3.0.2)
project(sidescan_slam)

## Compile as C++11, supported in ROS Kinetic and newer
add_compile_options(-std=c++11)

### GTSAM 
set(GTSAM_DIR /home/lin/develop/3rd/gtsam/install/lib/cmake/GTSAM)
find_package(GTSAM)
if(GTSAM_FOUND)  
  message(STATUS "Found GTSAM ${GTSAM_VERSION}")
else()
  message(STATUS "GTSAM Not Found")
endif()

find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
)

catkin_package()

include_directories(
  ${catkin_INCLUDE_DIRS}
  ${GTSAM_INCLUDE_DIR}
)

link_directories(
	${GTSAM_LIBRARY_DIRS}
)

### build test
add_executable(T_GTSAM_odomTest test/OdometryExample.cpp)
target_link_libraries(T_GTSAM_odomTest gtsam)
```
Then I run the executable, got the error:
>./T_GTSAM_odomTest: error while loading shared libraries: libmetis-gtsam.so: cannot open shared object file: No such file or directory

Any suggestions? Thank you so much!

acxz, 2020-07-09:
Hmm that is odd indeed. I got `OdometryExample` compiling as follows.

1. Install gtsam as specified in the [quick start](https://gtsam.org/get_started/)
2. Create the following structure:
```
test-gtsam
├── build
├── CMakeLists.txt
└── OdometryExample.cpp
```
3. Populate `OdometryExample.cpp` as an exact copy from https://github.com/borglab/gtsam/blob/develop/examples/OdometryExample.cpp
4. Populate `CMakeLists.txt` as follows:
```
project(OdometryExample)
cmake_minimum_required(VERSION 2.8)

find_package(GTSAM REQUIRED)

include_directories(${GTSAM_INCLUDE_DIR})

add_executable(OdometryExample OdometryExample.cpp)
target_link_libraries(OdometryExample gtsam)
```

5. Build and run the project
```
cd build
cmake ..
make
./OdometryExample
```

Please let us know if the following doesn't work for you. (Let's make sure we get OdometryExample working by itself before we try linking it with ROS catkin packages)

CrazyPopLin, 2020-07-09:
hi @acxz ,thanks for your reply. I did just like the **quick start** but installed the gtsam a self-defined path, so this time the CMakeLists.txt like below:
```s
project(test_gtsam)
cmake_minimum_required(VERSION 2.8)

### GTSAM 
set(GTSAM_DIR /home/lin/develop/3rd/gtsam/install/lib/cmake/GTSAM)
find_package(GTSAM)
if(GTSAM_FOUND)  
  message(STATUS "Found GTSAM ${GTSAM_VERSION}")
else()
  message(STATUS "GTSAM Not Found")
endif()

include_directories(${GTSAM_INCLUDE_DIR})

add_executable(OdometryExample OdometryExample.cpp)
target_link_libraries(OdometryExample gtsam)
```
Same issue occurred.  Maybe I used Ubuntu 20.04, because this system not tested according [Important Installation Notes](https://github.com/borglab/gtsam/blob/develop/INSTALL.md#important-installation-notes)?

acxz, 2020-07-09:
>  did just like the quick start but installed the gtsam a self-defined path

Can you give the exact commands you followed? Without them I am afraid I cannot reproduce your situation and therefore have limited knowledge in helping you.

> because this system not tested according Important Installation Notes?

That list sort of outdated anyways tbh. I'm running on Arch Linux with GCC 10.1.0. People that I know use GTSAM on Ubuntu 20.04 as well so it is not a system issue.

CrazyPopLin, 2020-07-09:
The install commands shown below:
```shell
$ cd <my_path>/
$ git clone https://github.com/borglab/gtsam
$ cd gtsam
$ mkdir build && mkdir install
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=<my_path>/gtsam/install ..
$ make check
$ install
```
This is new installed system, and it's first to install gtsam. Please let me know if you have any questions. thanks for your help.

acxz, 2020-07-09:
And how did you compile your own project?

CrazyPopLin, 2020-07-09:
[image]
The gtsam test project build steps are just like your suggestion.

acxz, 2020-07-09:
Weird those instructions shouldn't even compile your project if GTSAM is installed locally.

After installing GTSAM to a local directory, I built my project with the following commands:
```
cd build
cmake .. -DCMAKE_PREFIX_PATH=~/local
make
```

and just to iterate, where I installed GTSAM locally as follows after cloning:
```
cd gtsam
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX:PATH=$HOME/local
make install
```

With these steps I still do not receive the libmetis-gtsam issue.
Can you try these exact steps and let me know the output of them?

acxz, 2020-07-09:
>  The gtsam test project build steps are just like your suggestion.

Weird those instructions shouldn't even compile your project if GTSAM is installed locally.

After installing GTSAM to a local directory, I built my project with the following commands:
```
cd build
cmake .. -DCMAKE_PREFIX_PATH=~/local
make
```

and just to iterate, where I installed GTSAM locally as follows after cloning:
```
cd gtsam
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX:PATH=$HOME/local
make install
```

With these steps I still do not receive the libmetis-gtsam issue.
Can you try these exact steps and let me know the output of them?

CrazyPopLin, 2020-07-09:
## Build gtsam
```shell
$ cd 
$ mkdir local
$ cd local
$ git clone https://github.com/borglab/gtsam
$ cd gtsam
$ mkdir build 
$ cd build
$ cmake .. -DCMAKE_INSTALL_PREFIX:PATH=$HOME/local
$ make install
```
The CMake output for building gtsam:
>  -- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/home/lin/local/gtsam]
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.43") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found Eigen version: 3.3.7
-- Building 3rdparty
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building smart
-- Building navigation
-- GTSAM Version: 4.0.2
-- Install prefix: /home/lin/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.0.2
-- Install prefix: /home/lin/local
-- Wrote /home/lin/local/gtsam/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
[... 79 lines cut ...]
├── build
├── CMakeLists.txt
└── OdometryExample.cpp

#### CMakeLists.xtx
```shell
project(test_gtsam)
cmake_minimum_required(VERSION 2.8)

set(GTSAM_DIR /home/lin/local/lib/cmake/GTSAM)
find_package(GTSAM)
if(GTSAM_FOUND)  
  message(STATUS "Found GTSAM ${GTSAM_VERSION}")
else()
  message(STATUS "GTSAM Not Found")
endif()

include_directories(${GTSAM_INCLUDE_DIR})

add_executable(OdometryExample OdometryExample.cpp)
target_link_libraries(OdometryExample gtsam)
```
#### build project
```shell
```

ProfFan, 2020-07-10:
Before this is dealt with you can always use LD_LIBRARY_PATH to get going.

CrazyPopLin, 2020-07-10:
Hi @ProfFan , could you be more specific?

acxz, 2020-07-10:
@CrazyPopLin followed your steps command by command (except replacing `lin` with `acxz`) and I could not reproduce the issue. It seems there is something else amiss.

Can you give us the output of `ls ~/local/lib`?

It would be much appreciated if you can post your file as a [gist](https://gist.github.com) or as a [dropdown](https://gist.github.com/citrusui/07978f14b11adada364ff901e27c7f61) element in the issue comment.

CrazyPopLin, 2020-07-10:
### ls ~/local/lib
>cmake             libgtsam.so.4         libgtsam_unstable.so.4
libCppUnitLite.a  libgtsam.so.4.0.2     libgtsam_unstable.so.4.0.2
libgtsam.so       libgtsam_unstable.so  libmetis-gtsam.so

### install_manifest.txt
please check [here](https://gist.github.com/CrazyPopLin/7456c25ab82922795e9b5a2cd6f5223f).

Thanks.

CrazyPopLin, 2020-07-10:
I used the same build steps to install gtsam and build project in Jetson TX2 with jetpack 4.4.2(Ubuntu 18.04), got same error;
Then use same steps to install gtsam and build project in Ubuntu 16.04, it's working.
Just let you know.

ProfFan, 2020-07-10:
@CrazyPopLin I mean you can add `LD_LIBRARY_PATH` that points to where `libmetis-gtsam.so` is in and see if that fixes your issue.

Ref: https://stackoverflow.com/questions/13428910/how-to-set-the-environmental-variable-ld-library-path-in-linux

acxz, 2020-07-10:
@CrazyPopLin so it seems like that `libmetis-gtsam.so` is installed, but your executable cannot find it.

Glad to hear it's working on Ubuntu 16, but still very weird that it is not working for you on 18 or 20.

Can you change your CMakeLists.txt as follows and post the output, noting the new `get_target_property()` commands.

```
project(test_gtsam)
cmake_minimum_required(VERSION 2.8)

set(GTSAM_DIR /home/acxz/local/lib/cmake/GTSAM)
find_package(GTSAM)
if(GTSAM_FOUND)
  message(STATUS "Found GTSAM ${GTSAM_VERSION}")
else()
  message(STATUS "GTSAM Not Found")
endif()

include_directories(${GTSAM_INCLUDE_DIR})

add_executable(OdometryExample OdometryExample.cpp)
get_target_property(gtsam_link_lib gtsam INTERFACE_LINK_LIBRARIES)
message("${gtsam_link_lib}")
target_link_libraries(OdometryExample gtsam)
```

I get the following output:
```
-- GTSAM include directory:  /home/acxz/local/lib/cmake/GTSAM/../../../include
-- Found GTSAM 4.0.2
Boost::serialization;Boost::system;Boost::filesystem;Boost::thread;Boost::date_time;Boost::regex;Boost::timer;Boost::chrono;tbb;tbbmalloc;metis-gtsam
-- Configuring done
-- Generating done
-- Build files have been written to: /home/acxz/test-gtsam/build
```

CrazyPopLin, 2020-07-10:
@ProfFan ,thanks.
@acxz , the entire output:
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.43") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM include directory:  /home/lin/local/lib/cmake/GTSAM/../../../include
-- Found GTSAM 4.0.2
Boost::serialization;Boost::system;Boost::filesystem;Boost::thread;Boost::date_time;Boost::regex;Boost::timer;Boost::chrono;tbb;tbbmalloc;metis-gtsam
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lin/develop/dev/test_gtsam/build
```

CrazyPopLin, 2020-07-10:
@acxz , this issue is solved by addling following in ~/.bashrc suggested by @ProfFan 
`export LD_LIBRARY_PATH=/home/lin/local/lib/:$LD_LIBRARY_PATH`.

But this issue is wired, so please let me know if you want to debug more. thanks for your suggestion.

dellaert, 2020-07-10:
Thanks ! Will close for now

closed 2020-07-10 by dellaert

acxz, 2020-07-10:
I am puzzled but out of ideas. Nevetheless, glad to hear you got it working, and I hope you enjoy using GTSAM!

cynosure4sure, 2020-08-27:
Same issue when upgrading ROS Melodic code from 4.0.2 to 4.0.3 on Ubuntu 18.04. ROS code compiled correctly however at runtime gave error:
`error while loading shared libraries: libmetis-gtsam.so: cannot open shared object file: No such file or directory `

Adding `export LD_LIBRARY_PATH=/usr/local/lib/:$LD_LIBRARY_PATH` to ~/.bashrc resolved the issue as mentioned above.

Thank you for the solution.

bloom256, 2021-04-26:
Perhaps ldconfig should be called after installing gtsam. See https://www.debian.org/doc/debian-policy/ch-sharedlibs.html#s-ldconfig

cynosure4sure, 2021-05-03:
> Perhaps ldconfig should be called after installing gtsam. See https://www.debian.org/doc/debian-policy/ch-sharedlibs.html#s-ldconfig

Thank you this fixed the problem.

jingyibo123, 2021-09-18:
There's clearly a lot missing in the generated GTSAMConfig.cmake and GTSAM-exports.cmake

When installed to a custom path( which is usually the PREFERED way to build and link against a thirdparty project from source), the libgtsam.so is unable to find libmetis-gtsam.so (as seen by `ldd libgtsam.so`), and import GTSAM with find_package(GTSAM) is quite painful.

AFAIK, either the old fashioned way, define GTSAM_LIBRARIES and GTSAM_LIBRARY_DIR, or the modern way, define cmake targets with correct dependencies...

Correct me if I'm wrong I'm no cmake expert...

berndpfrommer, 2021-10-02:
When I try to fix this by creating a libgtsam4.triggers file in the debian directory:

`activate-noawait ldconfig`

 I get the following cryptic lintian error when packaging:
`E: libgtsam4: repeated-trigger-name activate-noawait ldconfig (line 3) vs activate-noawait ldconfig (line 1)
`
A web search didn't provide much clarity, but [the answer to this stack exchange post](https://unix.stackexchange.com/questions/519163/activate-noawait-ldconfig-trigger-for-runtime-library-package) mentions that the debian build tools add the trigger automatically, which may result in the above error message.

For now I will not include the ldconfig trigger also because it is not clear if it is still needed with the new package that uses the system-provided libmetis.

jlblancoc, 2021-10-03:
For the records: 
- Should #292 be updated to mark "libmetis" as solved? (Finally!! Yay! :+1: )
- I agree: `ldconfig` shouldn't be needed to be explicitly called in the Debian packaging, @berndpfrommer 

So good to see progress on #292 !

brunoeducsantos, 2021-12-03:
I am using Ubuntu 20.04 and my `cmake` is 
```
###############
### General ###
###############
cmake_minimum_required(VERSION 3.5.1)  # Ubuntu 16.04 (Xenial)

project(example)
    

if(CMAKE_COMPILER_IS_GNUCXX)
 set(CMAKE_CXX_FLAGS"-O2") ## Optimize
endif()

IF(MSVC)
  ADD_DEFINITIONS(-DNOMINMAX)
  # Another MSVC quirk. http://stackoverflow.com/questions/6832666/lnk2019-when-including-asio-headers-solution-generated-with-cmake
  SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /EHsc /MP") # For multi-threaded compilation on MSVC
ELSE()
  IF(APPLE)
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++") # C++11 support on Apple clang
  ELSE()
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")# C++11 support on GCC
  ENDIF()
ENDIF()
 set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
####################
### Dependencies ###
####################
find_package(Eigen3 REQUIRED)
find_package(PCL 1.8 REQUIRED)
find_package(grid_map_core REQUIRED)
find_package(grid_map_cv REQUIRED)
find_package(OpenCV REQUIRED )
find_library(localizationlib localization PATHS /usr/local/lib/)
find_library(mapslib maps PATHS /usr/local/lib/)
find_package(Qt5Widgets REQUIRED)
find_package(nanoflann)
FIND_PACKAGE(OpenMP)
find_package(GTSAM REQUIRED QUIET)
IF(OPENMP_FOUND)
  SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
  SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
  MESSAGE(STATUS "Found OpenMP")
ENDIF()

#############
### Build ###
#############
include_directories(
    include
    ${CMAKE_INCLUDE_DIR}
    ${EIGEN3_INCLUDE_DIRS}
    ${PCL_INCLUDE_DIRS}
    ${OpenCV_INCLUDE_DIRS}
    ${grid_map_core_INCLUDE_DIRS}
    ${grid_map_cv_INCLUDE_DIRS}
    ${lloc_hdmap_INCLUDE_DIRS}
    ${Qt5Widgets_INCLUDES}
    ${GTL_INCLUDES}
    ${GTSAM_INCLUDE_DIR}
)
add_definitions(${Qt5Widgets_DEFINITIONS})
set(CMAKE_CXX_FLAGS "${Qt5Widgets_EXECUTABLE_COMPILE_FLAGS}")
if (UNIX)
  add_definitions(-ggdb)
endif()
add_executable(${PROJECT_NAME} src/example.cpp)
target_link_libraries(${PROJECT_NAME} ${PCL_LIBRARIES}
${OpenCV_LIBS}
${OpenCV_LIBRARIES}
${grid_map_core_LIBRARIES}
${grid_map_cv_LIBRARIES}
${mapslib}
${processorlib}
${localizationlib}
${Qt5Widgets_LIBRARIES}
nanoflann::nanoflann
stdc++fs
OpenMP::OpenMP_CXX
gtsam
)
```
Getting an error like 

```
CMake Warning at CMakeLists.txt:66 (add_executable):
[cmake]   Cannot generate a safe runtime search path for target example because files
[cmake]   in some directories may conflict with libraries in implicit directories:
[cmake] 
[cmake]     runtime library [libgtsam.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
[cmake]       /usr/local/lib
```
which is causing a segmentation default.

Regards,
Bruno

ProfFan, 2021-12-03:
This is because you have *multiple* versions of GTSAM installed in different locations. Please search and remove them.

brunoeducsantos, 2021-12-03:
Thanks a lot ! Indeed that was the problem.

Janphr, 2022-02-27:
Unter 20.04 and GTSAM 4.2.0 following solved it for me:
`sudo cp /usr/local/lib/libmetis-gtsam.so /usr/lib`

wienans, 2023-01-20:
This works:
`sudo cp /usr/local/lib/libmetis-gtsam.so /usr/lib`
But is not so nice and not the intended way, for me also calling
`ldconfig`
worked fine on ubuntu 20.04