borglab/gtsam #398: Build errors after #386 was merged.

Wallbraker, 2020-07-12:
## Description

Change 95b4a49f643efe0a8bdae4c04132cf6b110deab2 causes build errors in Kimera-VIO. See CI build: http://ci-sparklab.mit.edu:8080/blue/organizations/jenkins/MIT-SPARK-Kimera/detail/PR-100/2/pipeline

```
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o:(.bss+0x10): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/frontend/StereoImuSyncPacket.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/frontend/StereoVisionFrontEnd.cpp.o:(.bss+0x10): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/frontend/VisionFrontEndModule.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/frontend/VisionFrontEndFactory.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/frontend/Tracker.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/backend/VioBackEndModule.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/backend/VioBackEnd.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/backend/RegularVioBackEnd.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/backend/RegularVioBackEndParams.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/mesh/Mesher.cpp.o:(.bss+0x20): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/mesh/MesherModule.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/mesh/MesherFactory.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/initial/InitializationFromImu.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/initial/InitializationBackEnd.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/initial/OnlineGravityAlignment.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/utils/ThreadsafeImuBuffer.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/pipeline/Pipeline.cpp.o:(.bss+0x18): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/pipeline/Pipeline-definitions.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetector.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/logging/Logger.cpp.o:(.bss+0x8): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/imu-frontend/ImuFrontEnd.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/imu-frontend/ImuFrontEndParams.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/visualizer/Visualizer3D.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/visualizer/Visualizer3DModule.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
CMakeFiles/kimera_vio.dir/src/visualizer/Visualizer3DFactory.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
[... cut ...]
CMakeFiles/kimera_vio.dir/src/visualizer/OpenCvDisplay.cpp.o:(.bss+0x0): multiple definition of `boost::archive::detail::extra_detail::init_guid<gtsam::PreintegrationCombinedParams>::g'
CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o:(.bss+0x0): first defined here
```

## Steps to reproduce

1. Build gtsam develop
2. Build Kimera-VIO

## Expected behavior

It builds

## Environment

Kimero-VIO CI picked it up, also breaks using Kimera-VIO docker build.

## Additional information

This was bisected using Kimera-VIO

Wallbraker, 2020-07-12:
Removing this line https://github.com/borglab/gtsam/blob/develop/gtsam/navigation/CombinedImuFactor.h#L388 fixes the build.

varunagrawal, 2020-07-12:
Are you importing the header in multiple places?

Wallbraker, 2020-07-12:
I am not familiar with the `Kimera-VIO` codebase at all but I would assume that they are doing this. That is kinda the point of a header, to include them at multiple places? The header has a `#pragma once` so it's something that is being defined in the header that emits a symbol that gets exported. No where in the gtsam code base is `BOOST_CLASS_EXPORT_GUID` used in a header, only `cpp` files.

varunagrawal, 2020-07-12:
You're correct. I did some research and turns out the `BOOST_CLASS_EXPORT_GUID` macro is fraught with issues, and have added a new fix. The PR should be merged in about an hour once CI passes. Thank you for reporting this.

varunagrawal, 2020-07-12:
@Wallbraker can you please try again with the latest develop branch?

closed 2020-07-12 by varunagrawal

Wallbraker, 2020-07-13:
@varunagrawal Yupp it works, thank you for taking care of this.