borglab/gtsam #405: MATLAB Toolbox Build Broken

mikesheffler, 2020-07-14:
The build of the MATLAB toolbox (at least on Windows) is currently broken. The errors are

```
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl gtsam::BayesNet<class gtsam::DiscreteConditional>::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (__imp_?print@?$BayesNet@VDiscreteConditional@gtsam@@@gtsam@@QEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z) referenced in function main	DiscreteBayesNetExample	C:\Mike\Work\Build\gtsam\examples\DiscreteBayesNetExample.obj	1
```

```

Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl gtsam::BayesNet<class gtsam::DiscreteConditional>::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (__imp_?print@?$BayesNet@VDiscreteConditional@gtsam@@@gtsam@@QEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z) referenced in function main	HMMExample	C:\Mike\Work\Build\gtsam\examples\HMMExample.obj	1	

```

```
Error	C2664	'int boost::mpl::assertion_failed<false>(boost::mpl::assert<false>::type)': cannot convert argument 1 from 'boost::mpl::failed ************boost::serialization::is_wrapper<T>::* ***********' to 'boost::mpl::assert<false>::type'	gtsam_matlab_wrapper	C:\local\boost_1_73_0\boost\archive\basic_xml_oarchive.hpp	89	
```

```
Error	C2789	'mpl_assertion_in_line_40': an object of const-qualified type must be initialized	gtsam_matlab_wrapper	C:\local\boost_1_73_0\boost\archive\basic_xml_oarchive.hpp	89	

```

The last two are weird and might be bogus -- I wouldn't be shocked if they went away after fixing the first two.

dellaert, 2020-07-14:
I think maybe this is because of #391. It might be solved by adding a specialized template for Matrix proper, so Matlab wrapper can latch on to it.

dellaert, 2020-07-14:
PS I'm trying that now @mikesheffler @varunagrawal

dellaert, 2020-07-14:
Although, I also think now it's BayesNet-inst.h that is not included. I will add it at end of BayesNet.h so it's always included.

dellaert, 2020-07-14:
@mikesheffler @varunagrawal I pushed a branch to https://github.com/borglab/gtsam/tree/fix/matlab_wrapper
It did *not* fix the issue, but hopefully you can pick up where I left off: I have to prepare my talk for today :-)

mikesheffler, 2020-07-14:
You know how in _Raiders of the Lost Ark_, Indiana Jones says, "Snakes! Why did it have to be snakes!?" This is like that, but with Boost Serialization. _Why did it have to be Boost Serialization!?_

Anyway, it appears that a `BOOST_SERIALIZATION_NVP` macro wrapper was missing for an element of `SOn.h` that was to be serialized. Locally, this builds. Can you guys take a quick look to make sure that makes sense to you (I don't know why it reared its head right now)? If this passes CI, it's probably good to go.

mikesheffler, 2020-07-14:
BTW: I accidentally pushed directly into `fix/matlab_wrapper` because I had named my previous fix the same thing, and my local GTSAM fork got confused about the upstream and the origin. Hopefully that's okay (it's just a one-liner).

dellaert, 2020-07-14:
Of course, I thought that was the plan all along :-)

PR #408 (2020-07-15): Fix/matlab wrapper
Hopefully this fixes #405 , thanks @mikesheffler
closed 2020-07-15 by dellaert