borglab/gtsam #406: gtsam::IndeterminantLinearSystemException on solving g2o file iteratively y isam2

victor1234, 2020-07-14:
Hello

I created sample which load `g2o` file, sort nodes and pass node one by one to ISAM2 solver. 

I loaded well known `parking-garage.g2o` sample.

And in _release_ compilation mode after hundreds of solving nodes `isam2 `produce error:
`gtsam::IndeterminantLinearSystemException`
in 
https://github.com/borglab/gtsam/blob/develop/gtsam/base/cholesky.cpp#L128

`A` matrix is 
```
     4.02877   -0.0154488   -0.0126849 -8.58045e-09 -5.15917e-09 -8.27612e-09      1.22758      2.77488     -2.60665            0            0            0
           0      4.12375    0.0216548  8.95375e-08   6.1106e-09  5.34852e-08     -2.96321     -1.02298     -2.48448            0            0            0
           0            0      4.13514  5.76561e-08 -4.38476e-08  2.35831e-09      2.37607      -2.7248     -1.71206            0            0            0
           0            0            0            1  1.34653e-14   1.5896e-14     -37963.2       281733       279130     0.310262     0.689857    -0.654091
           0            0    -0.637946     -7.06786            1   1.1391e-15  1.44476e+06 -1.41586e+06 -1.14019e+06    -0.740812    -0.255742    -0.621123
           0   -0.0306378     0.541343       3.3765     -0.13112            1  1.81628e+06       821609  1.21755e+06     0.595765    -0.677269    -0.431707
    -143.541      1.07851    0.0135804      -4.8176     0.205775     -0.24377  5.38763e+12 -5.64005e+11  5.53499e+11  1.07363e-10 -1.62578e+06 -1.65664e+06
     340.352     0.178699    -0.285675    -0.454868     0.498145  -0.00750081   -0.0103309  2.75908e+12  2.69334e+12  1.62578e+06 -3.78487e-11       340451
           0      2.42499    -0.270076     0.654105     0.384053  -0.00639875  0.000447994     0.270568  2.86037e+12  1.65664e+06      -340451 -6.24492e-12
           0            0      2.39522     -11.2022     0.131734    0.0167055   0.00444313    -0.477793  -0.00162247            1  5.56099e-17 -7.24618e-17
           0            0     -1.96279     -4.76188    -0.169352   0.00152002  -0.00896791   0.00827705    0.0113096     0.305683            1  5.36588e-17
           0     -1.31932      3.99788      5.87151    -0.166198   -0.0051872 -0.000647055    -0.152839    0.0126253    -0.111676    -0.169314            1
```

And in _debug_ mode:
https://pastebin.com/raw/G1Ntz9qf
additionally in _debug_ mode I obtain this output
https://pastebin.com/raw/JXzw9BRH

How to solve it?

**OS**
Ubuntu 20.04

**gtsam**
master branch

ProfFan, 2020-07-14:
Hi Victor, master is the prior release, could you try `develop` branch? Thanks a lot!

victor1234, 2020-07-14:
@ProfFan Sorry it was typo. I work in develop branch.
Fixed

varunagrawal, 2020-08-11:
This basically means the factor graph is singular, aka when the optimizer performs Cholesky decomposition on the formed sparse matrix, it fails because that matrix is singular.

Can you please provide a minimal working example of your code?

varunagrawal, 2021-01-02:
@victor1234 any updates? Please feel free to reopen if this is still an issue.

closed 2021-01-02 by varunagrawal