borglab/gtsam #407: [WIP] New MATLAB wrapper

ProfFan, 2020-07-14:
We need to finish this before 4.1.

- [ ] code generation for some method signatures is broken (wrong type)
- [ ] we have not tested on MATLAB at all (stuck in compilation)

ProfFan, 2020-07-14:
@dellaert

ProfFan, 2020-07-14:
I have updated the `feature/new_wrapper` branch to reflect the changes.

ProfFan, 2020-07-14:
I have finished the wrapping for matlab and briefly tested on MATLAB (loading and simple function works), could anyone help me test it by:

- Clone the `feature/new_wrapper` branch
- `git submodule update --init --recursive`
- mkdir build && cd build
- `cmake -DCMAKE_BUILD_TYPE=Release -DGTSAM_TYPEDEF_POINTS_TO_VECTORS=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON -DCMAKE_INSTALL_PREFIX=../gtsam_install -DGTSAM_CUSTOM_MATLAB_PATH=<your matlab>/MATLAB/R2020a/bin ..`
- `make -j8 install`
- Go to `../gtsam_install/gtsam_toolbox` and run the matlab wrapper

And report the issues you have? Many thanks!

@varunagrawal @mikesheffler @dellaert

ProfFan, 2020-07-14:
I am getting errors like this when running the examples:
```
PlanarSLAMExample
Error using gtsam.NonlinearFactorGraph
Error: File: NonlinearFactorGraph.m Line: 341 Column: 5
'push_back' is already defined as a method in the 'NonlinearFactorGraph' class.

Error in PlanarSLAMExample (line 29)
graph = NonlinearFactorGraph;
 ```

So we probably need to do some name mangling here.

ProfFan, 2020-07-14:
Same goes for the file name resolution. Really need some help from someone with better MATLAB experience...

varunagrawal, 2020-07-14:
I'll take a look at this later today.

ProfFan, 2020-07-14:
@varunagrawal Thank you in advance :) I need a list of problems so I can tweak the generation code...

ProfFan, 2020-07-16:
Another rather serious issue is that the current generator will generate identical signature for overloaded function, which is illegal in MATLAB:
```matlab
    function varargout = RzRyRx(varargin)
      % RZRYRX usage: RzRyRx(double x, double y, double z) : returns gtsam::Rot3
      % Doxygen can be found at http://research.cc.gatech.edu/borg/sites/edu.borg/html/index.html
      varargout{1} = gtsam_wrapper(263, varargin{:});
    end

    function varargout = RzRyRx(varargin)
      % RZRYRX usage: RzRyRx(Vector xyz) : returns gtsam::Rot3
      % Doxygen can be found at http://research.cc.gatech.edu/borg/sites/edu.borg/html/index.html
      varargout{1} = gtsam_wrapper(264, varargin{:});
    end
```

ProfFan, 2020-07-16:
Now most of the API is working, but construction of a simple factor crashes MATLAB:
```matlab
import gtsam.*

graph = NonlinearFactorGraph;

%% Add prior
priorNoise = noiseModel.Diagonal.Sigmas([0.3; 0.3; 0.1]);
graph.add(PriorFactorPose2(1, Pose2(0, 0, 0), priorNoise)); % Crashes here
```

I'll keep investigating what is happening, but any help is welcome...

dellaert, 2020-07-16:
I'm puzzled because many of these issues were *not* present for the c++ based wrap, and I thought we generated nearly identical c++ wrapper files. So, what's up with that? Let's talk in our meeting.

ProfFan, 2020-07-16:
@dellaert Sure, it's a lot of delicate name mangling differences....

ProfFan, 2020-07-16:
Now the crash is fixed and the examples mostly work.

ProfFan, 2020-07-22:
Since 4.0.3 is out, we now need testers for the new MATLAB wrapper, since the generated code currently does not exactly match what is generated by the old C++ wrapper.

dellaert, 2020-07-22:
> Since 4.0.3 is out, we now need testers for the new MATLAB wrapper, since the generated code currently does not exactly match what is generated by the old C++ wrapper.

It would be nice if this call came with instructions and a concrete ask :-) When fixed, also announce on internal slack?

ProfFan, 2020-07-22:
@dellaert Sure.

varunagrawal, 2021-03-18:
@ProfFan I checked the issue with `This` declarations and it seems to work fine. I believe we can close this issue. :slightly_smiling_face:

varunagrawal, 2021-03-18:
Nvm I encountered it again. I can fix it after I get the CI up and running again.

varunagrawal, 2021-03-18:
The `This` issue only happens when the class isn't templated, so that helps a lot.

varunagrawal, 2021-03-24:
Also, FYI I fixed all the Matlab tests so the tests now run and they pass.

varunagrawal, 2021-03-26:
This issue has been resolved :tada:

closed 2021-03-26 by varunagrawal