borglab/gtsam #413: Compilation of the Old C++ wrap fails in Release mode

ProfFan, 2020-07-16:
## Description

The old C++ wrap utility contains a use of uninitialized memory (which is probably caused by an undefined behavior)

## Steps to reproduce

1. Compile the wrap utility with GCC 10.1
2. Try to compile the MATLAB or Cython wrapper
3. Crash

## Expected behavior

Clean exit

## Environment

## Additional information

Valgrind:

```
==3944454== Memcheck, a memory error detector
==3944454== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==3944454== Using Valgrind-3.16.0.GIT and LibVEX; rerun with -h for copyright info
==3944454== Command: /home/fan/Projects/CV/SLAM/gtsam_second_build/wrap/wrap --matlab /home/fan/Projects/CV/SLAM/gtsam_second gtsam /home/fan/Projects/CV/SLAM/gtsam_second_build/gtsam/wrap/gtsam /home/fan/Projects/CV/SLAM/gtsam_second/gtsam
==3944454== 
==3944454== Memcheck, a memory error detector
==3944454== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==3944454== Using Valgrind-3.16.0.GIT and LibVEX; rerun with -h for copyright info
==3944454== Command: /home/fan/Projects/CV/SLAM/gtsam_second_build/wrap/wrap --matlab /home/fan/Projects/CV/SLAM/gtsam_second gtsam /home/fan/Projects/CV/SLAM/gtsam_second_build/gtsam/wrap/gtsam /home/fan/Projects/CV/SLAM/gtsam_second/gtsam
==3944454== 
==3944454== Use of uninitialised value of size 8
==3944454==    at 0x1B4A89: push_back (stl_vector.h:1189)
==3944454==    by 0x1B4A89: act<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char> > (push_back_actor.hpp:50)
==3944454==    by 0x1B4A89: operator()<char const*> (ref_const_ref_actor.hpp:70)
==3944454==    by 0x1B4A89: call<boost::spirit::classic::ref_const_ref_actor<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char>, boost::spirit::classic::push_back_action>, char const*> (scanner.hpp:147)
==3944454==    by 0x1B4A89: do_action<boost::spirit::classic::ref_const_ref_actor<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char>, boost::spirit::classic::push_back_action>, boost::spirit::classic::nil_t, char const*> (scanner.hpp:162)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:117)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (alternative.hpp:67)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (optional.hpp:56)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454== Use of uninitialised value of size 8
==3944454==    at 0x1B4A89: push_back (stl_vector.h:1189)
==3944454==    by 0x1B4A89: act<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char> > (push_back_actor.hpp:50)
==3944454==    by 0x1B4A89: operator()<char const*> (ref_const_ref_actor.hpp:70)
==3944454==    by 0x1B4A89: call<boost::spirit::classic::ref_const_ref_actor<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char>, boost::spirit::classic::push_back_action>, char const*> (scanner.hpp:147)
==3944454==    by 0x1B4A89: do_action<boost::spirit::classic::ref_const_ref_actor<std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char>, boost::spirit::classic::push_back_action>, boost::spirit::classic::nil_t, char const*> (scanner.hpp:162)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:117)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (alternative.hpp:67)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (optional.hpp:56)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (sequence.hpp:60)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: parse<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<> > > > (actions.hpp:113)
==3944454==    by 0x1B4A89: boost::spirit::classic::impl::concrete_parser<boost::spirit::classic::action<boost::spirit::classic::action<boost::spirit::classic::action<boost::spirit::classic::action<boost::spirit::classic::sequence<boost::spirit::classic::sequence<boost::spirit::classic::sequence<boost::spirit::classic::sequence<boost::spirit::classic::sequence<boost::spirit::classic::sequence<boost::spirit::classic::optional<boost::spirit::classic::alternative<boost::spirit::classic::action<wra [... cut ...] t::spirit::classic::iteration_policy>, boost::spirit::classic::match_policy, boost::spirit::classic::action_policy> >, boost::spirit::classic::nil_t, boost::spirit::classic::nil_t> >, boost::spirit::classic::rule<boost::spirit::classic::scanner<char const*, boost::spirit::classic::scanner_policies<boost::spirit::classic::skipper_iteration_policy<boost::spirit::classic::iteration_policy>, boost::spirit::classic::match_policy, boost::spirit::classic::action_policy> >, boost::spirit::classic::nil_t

ProfFan, 2020-07-16:
Asan fails to detect this issue because the memory gets properly populated once I enable sanitizers... But valgrind's result is not point-on enough to pin-point the issue.
@jlblancoc @varunagrawal Any ideas?

varunagrawal, 2020-07-21:
Yeah I am having this same issue and the weird part is that it works with Debug mode.
I tried using an older commit to see if it was something we introduced, and no, the issue still persists. This makes me think that something with GCC or Boost is the root cause.

ProfFan, 2020-07-21:
Yep, I switched to Clang and no problem whatsoever. Could be GCC whose messed up :)

varunagrawal, 2020-07-21:
Interesting. I am also using an Ubuntu machine and the latest commit compiles perfectly. The issue seems to be on Arch primarily. Maybe it is an issue with the latest version of GCC. :/

acxz, 2020-07-25:
Just as another data point, `testWrap` and `testClass` segfault for me at the same root cause of the error. (GCC compiled, Clang compiled the tests pass)

varunagrawal, 2020-07-28:
This definitely seems to be a GCC issue. Given that we are focused on pushing the pybind wrapper forward in favor of the cython wrapper, this issue can be deemed as low-priority.

dellaert, 2020-07-28:
If someone fixes it, it would be good to do so now so we can release another maintenance version *right* before 4.1, which will delete cython. We already accumulated 10 PRs or so after 4.0.3 release...

varunagrawal, 2021-09-19:
Closing this since we're pushing the Pybind wrapper

closed 2021-09-19 by varunagrawal