borglab/gtsam #415: [Question] question about the derivatives of LieGroup

mklingen-bd, 2020-07-20:
It's difficult to know what the intention is here without documentation , but I was surprised to see the code for `Pose3::compose` looks like this:

```C++
  Class compose(const Class& g, ChartJacobian H1,
      ChartJacobian H2 = boost::none) const {
    if (H1) *H1 = g.inverse().AdjointMap();
    if (H2) *H2 = Eigen::Matrix<double, N, N>::Identity();
    return derived() * g;
  }
```

I was under the impression that if we have `f(x) = A.compose(x)`, then:

```
H1 = dfdA
H2 = dfdx
```

But it looks like H2 is always the identity. So how do I actually obtain `dfDX`?

dellaert, 2020-07-20:
Look at the documentation, which is in math.pdf. Bottom-line: these Jacobians compute the correct Lie algebra "vectors" for how we define retract, which is in body coordinates, not global coordinates.
If still questions, consider posting in Google Group :-)

mklingen-bd, 2020-07-20:
Let me rephrase this as an issue: "The documentation is inconsistent and confusing."

Obviously I'm sure your math is correct, and the math pdf helps clarify what is meant by these functions in a theoretical sense -- but the variable names are both inconsistent and confusing, which makes interfacing with this library a challenge.

Take this example in gtsam "concepts" from the documentation on Lie groups:

```c++
traits<T>::Compose(p,q,Hq,Hp)
```

This differs from the actual implementation in c++:

```c++
  static Class Compose(const Class& m1, const Class& m2, //
      ChartJacobian H1 = boost::none, ChartJacobian H2 = boost::none) {
    return m1.compose(m2, H1, H2);
  };
```

It's unclear how p, q, g, Hq, Hp, m1, m2, H1, and H2 relate to each other. Is H1 the derivative of `Compose` w.r.t m1? If so, why does the gtsam "concepts" documentation list this as "Hq?" It would be helpful if the variables had clearer names and/or explanatory comments.

dellaert, 2020-07-20:
Yeah, `Compose(p,q,Hq,Hp)` is definitely a typo: the arguments are reversed: Hq is the derivative with respect to q and Hp is with respect to p, and we always preserve the order. So, the documentation *is* not only confusing but wrong in this case :-)

Do you want to do a PR or should I :-) ?

dellaert, 2020-07-21:
Added PR #416 to make OptionalJacobian more consistent with [GTSAM-Concepts.md](GTSAM-Concepts.md])

jlblancoc, 2020-07-21:
> Let me rephrase this as an issue: "The documentation is inconsistent and confusing."
> ...
>
>     traits<T>::Compose(p,q,Hq,Hp)
>                             ^  ^

IMHO, I would call that a minor typo instead... ;-) 
But I understand it may cause frustration.

dellaert, 2020-07-23:
Solved for Pose3 in #416

closed 2020-07-23 by dellaert