borglab/gtsam #421: references

xdtl, 2020-07-22:
## Feature

reference list

## Motivation

In the code I saw comments such as: refer to reference [2]. Can I have the reference list somewhere?

## Pitch

Add a link to the list of references.

jlblancoc, 2020-07-22:
BTW: Doxygen supports citations from a .bibtex file. Could be done that way?

acxz, 2020-07-23:
> In the code I saw comments such as: refer to reference [2]. Can I have the reference list somewhere? 

Can you explicitly link to where this occurs?

> BTW: Doxygen supports citations from a .bibtex file. Could be done that way?

o.O that's something neat I didn't know

xdtl, 2020-07-23:
Thank you both for your replies!

I saw them in gtsam\gtsam\navigation\CombinedImuFactor.cpp, line 74-78. Now I actually see the reference list at the beginning of CombinedImuFactor.h. That's all I need. Thank you for the great work!

acxz, 2020-07-23:
@xdtl Glad to see the reference was there! You can go ahead and close the issue if you got it figured out.

closed 2020-07-24 by varunagrawal