borglab/gtsam #433: AdjointMap jacobians

gchenfc, 2020-07-25:
## Feature

Lie Groups have `AdjointMap` implemented.

Just as (lowercase) `Pose3.adjointMap` has functions for
```c++
Matrix66 adjointMap() { ... }
Vector6 adjoint(Vector6 y) { ... }
Vector6 adjoint(Vector6 y, OptionalJacobian H) { ... }
```

It would be useful to have the equivalent one with jacobians for (uppercase) `AdjointMap`.  Also, I propose to add another argument:

```c++
Vector Adjoint(Vector6 y, OptionalJacobian H, OptionalJacobian H_y) { ... }
```

to avoid having to calculate the AdjointMap again.

Unless there is a reason against this, I can begin working on this.

## Motivation

Factors which need to convert tangent vectors to different frames will need the derivatives of AdjointMap.  For example, in GTDynamics many factors such as TwistFactor, TwistAccelFactor, and WrenchEquivalenceFactor need this.  Currently, GTDynamics is using an ugly way that would become much cleaner if this were implemented, at least for Pose3.

## Notes

Currently only Pose3 has a function `Adjoint(Vector6 y) { return AdjointMap() * y; }`, and also Pose3's `AdjointMap` has a comment " /// FIXME Not tested - marked as incorrect".

dellaert, 2020-07-25:
Sure. Make sure Jacobians for methods are ordered `HSelf, Hargument`. If this means the API **changes**, we need to think a bit how to handle, e.g., force the use of the two-argument jacobian version by not having defaults.

ProfFan, 2021-07-19:
Ping @gchenfc

PR #885 (2021-11-01): `Pose3::Adjoint(xi)` Jacobians
Closes #433 

This PR implements the Jacobians for `Pose3::Adjoint(x)` := `Pose3::AdjointMap() * x`.

It uses `Rot3.rotate` and `cross(Point3, Point3)` to calculate intermediate derivatives.

This should help in GTDynamics to eliminate the `AdjointMapJacobianQ` function and allow it to be generalized to other joint types.
closed 2021-11-01 by gchenfc