borglab/gtsam #439: Separate Pybind modules for inner namespaces

varunagrawal, 2020-07-28:
## Feature

Currently, all namespaces under the top level `gtsam` (e.g. `symbol_shorthand`) are added as pybind submodules. This requires the python import to look like 

```python
from gtsam.gtsam.symbol_shorthand import X
```

aka a gtsam module inside a gtsam module.

We can get rid of the inner gtsam module by configuring the pybind wrapper to create separate modules for all the inner namespaces, allowing us to do

```python
from gtsam.symbol_shorthand import X
```

which is more intuitive and neater, especially since `import gtsam.symbol_shorthand` works.

## Motivation

Outlined above.

## Pitch

Update the Pybind wrapper to have separate modules for each inner namespace.
Currently there are 5:

- [ ] noiseModel
- [ ] mEstimator
- [ ] symbol_shorthand
- [ ] imuBias
- [ ] utilities

## Alternatives

N/A

## Additional context

@ProfFan should be able to provide details on whether this is doable without breaking other stuff since he's the lead on the Pybind wrapper. :-)

ProfFan, 2020-07-28:
It's just time consuming :shrug: I feel it's more like a cosmetic issue (and I feel that it may be possible to solve with a trick)

[spoiler]

add a `symbol_shorthands` subfolder, put a `__init__.py` there, and add delegates.

dellaert, 2020-07-28:
I do agree with @varunagrawal there is an issue here. @ProfFan cosmetics *matters* for our users.
Also, "adding delegates" may be something that adds maintenance issues? 
Typing is not the bottleneck here so I'd like to avoid tricks if possible.

ProfFan, 2020-07-28:
@dellaert It's because we are currently having this one massive header and it's not trivial to escape this design pattern, as the wrap script is designed for this. Maybe in the future we can split the headers so this won't be an issue, but for now, I don't see an easy way to get this working.

varunagrawal, 2020-07-28:
So I actually did quite a bit of research on this today. I'd be glad to put in a PR for this once I'm done with my summer commitments.

dellaert, 2020-07-28:
Could you write up the solution in an issue? Maybe with links?

dellaert, 2020-07-28:
Or, *this* issue, of course :-)