borglab/gtsam #447: Replace Point-Vector typedef with empty derived class

varunagrawal, 2020-07-29:
## Feature

Instead of this

```cpp
typedef Vector3 Point3;
```

I propose using this:

```cpp
class Point3 : public Vector3 {};
```

## Motivation

If we `typedef` the PointX classes, we not only get different print behavior (amongst other things), we also have to account for this in multiple places, littering the codebase with `#ifdef GTSAM_TYPEDEF_POINTS_TO_VECTORS` guards in places like tests and other classes.

By using the empty derived class structure above, thanks to the __zero-cost abstraction__, we can use the PointX classes more naturally without having to account for difference in subtle behavior, while enjoying all the lean benefits of the typedef.

## Pitch

Update the PointX classes so that the conditional compilation happens only inside the class definition.

```cpp
class Point3 : public Vector3 {

#ifndef GTSAM_TYPEDEF_POINTS_TO_VECTORS
// ... all the code if Points are not typedef'd to vectors
#endif

};
```

This is to be followed by updating the codebase in general to remove all the extra guards etc.

## Alternatives

Keep things as they are.

## Additional context

ProfFan, 2020-07-29:
Eigen does not recommend subclassing, but YMMV

dellaert, 2020-07-29:
No, sorry. In fact I want to totally get rid of Point2, Point3... I think only holdup is matlab wrapper.

varunagrawal, 2020-07-29:
Coolio

closed 2020-07-29 by varunagrawal

jlblancoc, 2020-07-29:
> No, sorry. In fact I want to totally get rid of Point2, Point3... I think only holdup is matlab wrapper.

@dellaert I'm curious about why do you want to remove them, and what would be the alternatives, since I'm using those types in other projects... :-)

ProfFan, 2020-07-29:
I think "get rid of" means typealiasing? Since they are indeed just Vector2 and Vector3, and the Eigen version can use auto vectorization while the GTSAM-defined version cannot.

dellaert, 2020-07-29:
If we delete the class and just keep eigen, will MATLAB work? If so, we should do it in 4.1 release.

ProfFan, 2020-07-29:
@dellaert I think we discussed it last time (on the context of building python and matlab together), but you suggested that we make it work first so I didn't push that way. It's definitely doable though, I will look into it.

varunagrawal, 2020-07-29:
So my gripe with this is that I want to keep having Point2 and Point3 print like 
```
[x, y, z]'
```

instead of 
```
x
y
z
```
which is what Eigen defaults to.

One way of accomplishing this is to overload the `<<` operator, but then this affects the Vector3 implementation directly, having consequences for streams where Vector3 is used (and not Point3), such as in saving Logmap representations. This necessitates the need to specially handle those usecases (which is painful and error prone).

The empty derived class seemed like a good compromise since with next to no overhead, we would be able to merge the functionality of the two and still have a clean separation and customization.

varunagrawal, 2020-07-29:
@ProfFan thanks for the note about Eigen subclassing. Instead of subclassing, we should simply use [Eigen's plugin system](https://eigen.tuxfamily.org/dox/TopicCustomizing_Plugins.html).

jlblancoc, 2020-07-29:
I strongly recommend against the plug in mechanism, after using it in another project for years. It's a nightmare since all dependent projects must ensure including your headers first, always...

ProfFan, 2020-07-29:
Yeah, anything that tweaks Eigen is a nightmare, if we ignore Eigen itself as a nightmare...

jlblancoc, 2020-07-29:
Hahaha, absolutely! And a sink of build time...

ProfFan, 2020-07-29:
@dellaert It works, we can discuss in our meeting, still need some tweak in the generated MATLAB type check code but it looks good.
```matlab
>> gtsam.Pose3(gtsam.Rot3(), [0 0 0]')
R:
[
	1, 0, 0;
 	0, 1, 0;
 	0, 0, 1
]
0
0
0;
>> gtsam.Pose3(gtsam.Rot3(), [0 0 1]')
R:
[
	1, 0, 0;
 	0, 1, 0;
 	0, 0, 1
]
0
0
1;
>> p3=gtsam.Pose3(gtsam.Rot3(), [0 0 1]')
R:
[
	1, 0, 0;
 	0, 1, 0;
 	0, 0, 1
]
0
0
1;
>> p3.translation

ans =

     0
     0
     1

>> nlfg = gtsam.NonlinearFactorGraph()
size: 0
>> nlfg.addPriorPoint3(0, [0 0 0]', gtsam.noiseModel.Unit.Create(3))
>> nlfg
size: 1

Factor 0: PriorFactor on 0
  prior mean: [
	0;
 	0;
 	0
]
  noise model: unit (3) 
```

dellaert, 2020-07-29:
> > No, sorry. In fact I want to totally get rid of Point2, Point3... I think only holdup is matlab wrapper.
> 
> @dellaert I'm curious about why do you want to remove them, and what would be the alternatives, since I'm using those types in other projects... :-)

They are typedef'd to Vector2 and Vector3, anyway. we'll keep the typedefs, of course, just not the old implementation, if we can make matlab works (which looks promising).