borglab/gtsam #448: Community Forum Update

acxz, 2020-07-29:
## Feature

This is a feature request for moving the discussion from the GTSAM [Google Group](https://groups.google.com/forum/#!forum/gtsam-users) to a more modern Community Forum solution, specifically [Discourse](https://www.discourse.org/).

## Motivation

Google groups is an older service and there are many things to be desired.
The biggest issue is the fact that you must sign in with a Google account.
Discourse allows you to sign in with multiple options, including Github login.

Google groups is not open source, whereas Discourse is.

Searching is pretty painful. This is not an issue on Discourse.

Discourse has emojis!

Can use categories, tags, and discourse comes with built in moderation tools.

All in all Discourse offers all the features in Google Groups and much more.

## Pitch

Move Google Group over to a GTSAM Discourse.

1. Sign up for a Discourse for GTSAM: Discourse offers free hosting for open source projects, see [here](https://blog.discourse.org/2018/11/free-hosting-for-open-source-v2/)
2. Import Google Groups discussion over to Discourse, see [here](https://meta.discourse.org/t/importing-mailing-lists-mbox-listserv-google-groups-emails/79773)
3. Shutdown old Google Groups and start using the new Discourse.

## Alternatives

I think Discourse is the best one for the job.

## Additional context

I really do think this will further facilitate GTSAM discussions.

dellaert, 2020-07-29:
I'm willing to try it: I did step 1, if you could do step 2 :-)

acxz, 2020-07-29:
As for step 2, I believe someone who is a Google Groups owner privileges needs to handle it.

From import guide:
> In order to download messages, the crawler needs to have access to a Google account that has the owner role for your group.

I could handle step 2, if I become an owner for the Google Group.

After the import is done on a local discourse server, we can ask Discourse folks to restore that local discourse server/backup and put it up on their hosting site.

From free hosting discourse blog
> If you handle the import yourself, we’re happy to restore a Discourse backup of your import site onto our hosting at no additional charge.

dellaert, 2020-08-02:
@acxz I have the discourse group. Send me email on GIT side with credentials you want to use for Google group

acxz, 2020-08-02:
Just sent a request to join the gtsam user groups under display name acxz as well as sent my email I used to sign up from my GT email address.

varunagrawal, 2020-09-11:
@acxz any updates on this? I would very much like to use Discourse over Google Groups.

acxz, 2020-09-11:
Sorry, wasn't able to create the local discourse server with the google groups data before the semester started, and now I'm swamped with homework and research...

ProfFan, 2021-07-19:
Maybe we should try GitHub discussions? @varunagrawal @dellaert

varunagrawal, 2021-07-19:
Is that a new GitHub feature? Anything that doesn't require too much manual effort (like discourse) and that doesn't need me to always use my Gmail to login would be great.

ProfFan, 2021-07-19:
https://github.com/borglab/gtsam/discussions I just enabled it. Next is to add a BIG RED NOTICE on the readme so people don't post questions in issues. :robot:

varunagrawal, 2021-08-01:
We can close this since @ProfFan has enabled Github Discussions on this repo.

closed 2021-08-01 by varunagrawal