borglab/gtsam #450: Add gtsam to vcpkg

talregev, 2020-07-30:
> ## Feature

Add gtsam to vcpkg
> 

> ## Motivation

Easier to compile gtsam to windows. there infrastructure with all the 3rparty that gtsam need, and can add other libraries as well.
> 

> ## Pitch

when I type vcpkg install gtsam it install the gtsam in windows.
> 

## Alternatives

## Additional context

varunagrawal, 2020-07-30:
Hi @talregev. This seems like a cool idea, but I don't think any of us have the bandwidth to accomplish this any time soon.

Taking a quick look, packaging GTSAM via the github repo is straightforward ([link](https://vcpkg.readthedocs.io/en/latest/examples/packaging-github-repos/)). If you could make a PR for this, I'd be glad to help review and merge it in.

talregev, 2023-04-07:
fixed in here:
https://github.com/microsoft/vcpkg/pull/30560

closed 2023-04-07 by talregev